linuxboot/heads-wiki #33: Build breaks

aroobam, 2019-11-06:
Hi team,
Looks like something broke with musl-linux gcc. Says "command not found". See partial build dump below,
I am on the same system that was working earlier. Just tried a fresh git clone and make.

You can now run IASL ACPI compiler from /home/madhan/heads/build/coreboot-4.8.1/util/crossgcc/xgcc.
make[2]: Leaving directory '/home/madhan/heads/build/coreboot-4.8.1'
#echo '******* Building crossgcc-arm (this might take a while) ******'
#/home/madhan/heads/build/make-4.2.1/make -C "/home/madhan/heads/build/coreboot-4.8.1" crossgcc-arm
2019-11-06 13:14:15+05:30 CONFIG coreboot
git clone https://github.com/GregorR/musl-cross "/home/madhan/heads/build/musl-cross-git"
Cloning into '/home/madhan/heads/build/musl-cross-git'...
remote: Enumerating objects: 129, done.
remote: Counting objects: 100% (129/129), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 925 (delta 58), reused 118 (delta 50), pack-reused 796
Receiving objects: 100% (925/925), 342.27 KiB | 490.00 KiB/s, done.
Resolving deltas: 100% (557/557), done.
cd /home/madhan/heads/build/musl-cross-git && git submodule update --init --checkout
if [ -r patches/musl-cross.patch ]; then ( cd /home/madhan/heads/build/musl-cross-git ; patch -p1 ) < patches/musl-cross.patch || exit 1 ; fi
patching file config.sh
if [ -d patches/musl-cross ] && [ -r patches/musl-cross ] ; then for patch in patches/musl-cross/*.patch ; do echo "Applying patch file : $patch " ; ( cd /home/madhan/heads/build/musl-cross-git ; patch -p1 ) < $patch || exit 1 ; done ; fi
2019-11-06 13:14:25+05:30 CONFIG musl-cross
2019-11-06 13:14:25+05:30 MAKE musl-cross
2019-11-06 13:40:13+05:30 DONE musl-cross
2019-11-06 13:40:14+05:30 WGET https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.14.62.tar.xz
--2019-11-06 13:40:14--  https://cdn.kernel.org/pub/linux/kernel/v4.x/linux-4.14.62.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 151.101.193.176, 151.101.129.176, 151.101.65.176, ...
Connecting to cdn.kernel.org (cdn.kernel.org)|151.101.193.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 100957512 (96M) [application/x-xz]
Saving to: '/home/madhan/heads/packages/linux-4.14.62.tar.xz.tmp'

/home/madhan/heads/ 100%[===================>]  96.28M  1.68MB/s    in 77s     

2019-11-06 13:41:33 (1.25 MB/s) - '/home/madhan/heads/packages/linux-4.14.62.tar.xz.tmp' saved [100957512/100957512]

echo "51ca4d7e8ee156dc0f19bc7768915cfae41dbb0b4f251e4fa8b178c5674c22ab  /home/madhan/heads/packages/linux-4.14.62.tar.xz" | sha256sum --check -
/home/madhan/heads/packages/linux-4.14.62.tar.xz: OK
tar -xf "/home/madhan/heads/packages/linux-4.14.62.tar.xz" -C "/home/madhan/heads/build"
if [ -r patches/linux-4.14.62.patch ]; then ( cd /home/madhan/heads/build/linux-4.14.62 ; patch -p1 ) < patches/linux-4.14.62.patch || exit 1 ; fi
if [ -d patches/linux-4.14.62 ] && [ -r patches/linux-4.14.62 ] ; then for patch in patches/linux-4.14.62/*.patch ; do echo "Applying patch file : $patch " ; ( cd /home/madhan/heads/build/linux-4.14.62 ; patch -p1 ) < $patch || exit 1 ; done ; fi
Applying patch file : patches/linux-4.14.62/0000-efi_bds.patch 
patching file arch/x86/boot/compressed/eboot.c
Applying patch file : patches/linux-4.14.62/0001-fake-acpi.patch 
patching file drivers/acpi/acpica/evxfevnt.c
patching file drivers/acpi/acpica/hwacpi.c
Applying patch file : patches/linux-4.14.62/0002-nmi-squelch.patch 
patching file arch/x86/kernel/nmi.c
Hunk #1 succeeded at 292 (offset -11 lines).
Hunk #2 succeeded at 302 (offset -11 lines).
Applying patch file : patches/linux-4.14.62/0003-fake-trampoline.patch 
patching file arch/x86/realmode/init.c
Hunk #1 succeeded at 37 (offset 2 lines).
Hunk #2 succeeded at 151 (offset 13 lines).
Applying patch file : patches/linux-4.14.62/0010-winterfell-ahci.patch 
patching file drivers/ata/libahci.c
Hunk #1 succeeded at 543 (offset 6 lines).
2019-11-06 13:42:14+05:30 CONFIG linux
2019-11-06 13:42:23+05:30 MAKE linux
tail /home/madhan/heads/build/log/linux.log
-----
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_x32.h
make[4]: /home/madhan/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
  HOSTCC  scripts/basic/bin2c
make[4]: /home/madhan/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  arch/x86/tools/relocs
make[4]: /home/madhan/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
make[4]: /home/madhan/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
  Using .. as source for kernel
make[4]: /home/madhan/heads/build/../crossgcc/x86_64-linux-musl/bin/x86_64-musl-linux-gcc: Command not found
Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make[4]: *** [/home/madhan/heads/build/linux-4.14.62/Makefile:1110: prepare-compiler-check] Error 1
make[3]: *** [Makefile:146: sub-make] Error 2
make[2]: *** [Makefile:24: __sub-make] Error 2
make[2]: Leaving directory '/home/madhan/heads/build/linux-4.14.62/linux-qemu'
make[1]: *** [Makefile:378: /home/madhan/heads/build/linux-4.14.62/linux-qemu/.build] Error 1
make[1]: Leaving directory '/home/madhan/heads'
Makefile:584: recipe for target 'all' failed
make: *** [all] Error 2

aroobam, 2019-11-14:
Hi Team,
I found what was breaking. The linux-musl output file names are mixed up. The make generates files heads/crossgcc/x86_linux-musl/bin/*-linux-musl-* files, but elsewhere it refers to .... *-musl-linux-* during the make. The earlier builds I ran has soft links in the required names. The latest build process does not generate these soft links and hence the build break. I created the links manually and restarted the build. It picked up from where it broke and completed successfully.

I just tried a fresh git clone and checked. Still it is broken and needs to be fixed. Can someone please take a look on what changed or at least let me know if I am wrong and missing something so that we can close this issue one way or another?

Thanks,
Madhan.

tlaurion, 2019-11-14:
https://github.com/osresearch/heads/pull/618

closed 2020-08-01 by tlaurion