linuxboot/heads-wiki #35: Update documents to mention software only method.

YellowOnion, 2020-01-23:
There's a method here for flashing firmware using an exploit in earlier versions of Lenovo bios'

https://github.com/gch1p/thinkpad-bios-software-flashing-guide

I'm working on a live CD (NixOS base) with all the tools needed which will be GPL'd when necessary.

tlaurion, 2020-01-23:
@YellowOnion : keep in mind that the IFD (descriptor)[ still needs to be flashed externally](https://nlnet.nl/project/AccessibleSecurity/) to change [ME neutered+deactivated partition size to use freed ME 5mb available space](https://github.com/osresearch/heads/issues/307#issuecomment-547163607) by distibuting the expended IFD regions and trimmed ME binaries, [since we are running out of space to add more tools](https://github.com/osresearch/heads/issues/590).

IFD (descriptor), in that case, would still need to be [externally flashed](https://gitlab.com/tlaurion/heads/blob/PrivacyBeast_X230_QubesOS_Certified_thin-provisioning-tools/boards/x230-libremkey/x230-libremkey.config#L44). The 12mb coreboot image in last link being splitted out in two externally flashable roms (top and bottom rom images) seems to be the way to go.

SebastianMcMillan, 2020-01-23:
I'd like to also add that a LiveUSB of said tools already exists at https://github.com/n4ru/1vyrain/

YellowOnion, 2020-01-23:
@SebastianMcMillan 1vyrain's script is not GPL, the author has a feud with r/thinkpad and files DCMA to anyone linking it [on the reddit](https://www.reddit.com/r/thinkpad/comments/er30pz/1vyrain_not_allowed/), The live USB is not "open", contains copyrighted content (the modded lenovo bios), I don't know how he made it, or how to reproduce it, and it's compressed using a proprietary format (rar), and is a massive 3.6GB, my NixOS based CD has most of the tools so far, and is only 250MB, will be reproducible by anyone, will not include proprietary code.

digmorepaka, 2020-02-22:
@YellowOnion https://medium.com/@n4ru/1vyrain-self-build-documentation-5059825b1fdb

1. 1vyrain script is public at https://github.com/n4ru/1vyrain

2. The live USB image is a 257MB Debian live usb in iso format

3. You can go extract your own bios region image from Lenovo's update package and use UEFIPatch

YellowOnion, 2020-02-22:
> @YellowOnion https://medium.com/@n4ru/1vyrain-self-build-documentation-5059825b1fdb
>
> 1. 1vyrain script is public at https://github.com/n4ru/1vyrain
> 2. The live USB image is a 257MB Debian live usb [...]

> License
> I retain all rights to the code found in this repo (excluding the BIOS roms and the patcher binary), and no one may reproduce, distribute, or create derivative works from this repo without including this README.me in its entirety!

> This project is not permitted to be shared on the ThinkPad subreddit.

The Live CD was not, when I made the comment public, it is still not public, it was a big huge rar compressed image for a 4GB USB, the script for flashing does not build the Live CD, we cannot reproduce the the image exactly because of this, the script is copyrighted under a non-free licence (as quoted above).

Please check dates a lot can happen in a month.

And please try to infer my point, a project like heads is not meant to rely on proprietary code, dodgy isos from an unknown individual, especially when that individual has a history of using DMCA to control information.

tlaurion, 2020-08-01:
Unfortunately, it is impossible to unlock IFD externally. This needs SPI flash to be backuped, have ifdtool unlock ther IFD inside of the ROM image unlocked and flashed back. While this boot disk can reflash and neuter already defined regions inside of the IFD, it cannot redefine those regions inside of IFD flash descriptor.

That was linked to [external project issue and closed](https://github.com/n4ru/1vyrain/issues/2)

closed 2020-08-01 by tlaurion

tlaurion, 2020-08-01:
@YellowOnion feel free to tag me if you think this issue was closed where it should have stayed open.