linuxboot/heads-wiki #36: Update Heads repo with librem script for x230 and debian install

skollrc, 2020-01-27:
Hi all, 

Purism did a great job to simplify heads build and install. Is it possible for the project to update doc and script with librem's one? 
The current doc is hard to understand (where to start, how to properly install debian based system, how to flash with existing coreboot rom...)

Let me know if I can help in any way :-)

MrChromebox, 2020-01-27:
which script are you referring to?

much of our (Purism's) documentation is Librem-centric, and would need some measure of generalization before it would be appropriate for use for the Heads project as a whole

skollrc, 2020-01-27:
https://source.puri.sm/coreboot/utility/raw/master/coreboot_util.sh this one
the doc is this one: https://docs.puri.sm/PureBoot/Heads/User_Manual.html#heads-beta-installation-instructions 

Building Heads is, finally, not that hard but the documentation is hard to understand (and how heads works precisely).

MrChromebox, 2020-01-27:
> https://source.puri.sm/coreboot/utility/raw/master/coreboot_util.sh this one

that script (which I wrote) doesn't build Heads. It allows Librem users to flash precompiled builds of our Heads fork (PureBoot) or coreboot/SeaBIOS, or to build coreboot/SeaBIOS from source, but does not (yet) offer the option of building Heads (PureBoot) from source.

skollrc, 2020-01-27:
ok, and is there any way to have prebuild heads images (including coreboot and cleaned ME) for x230? 

I didn't understand everything yet but maybe I can help for x230 (including rewrite a doc)

MrChromebox, 2020-01-27:
> ok, and is there any way to have prebuild heads images (including coreboot and cleaned ME) for x230?

there's nothing legally or logistically preventing one from having prebuilt Heads images, though things get murky when you add non-redistributable binaries like the ME into the mix. Currently, the Heads project itself doesn't distribute precompiled/binary images for any boards/devices.

skollrc, 2020-01-27:
Ok, thx for your answers :-)

closed 2020-01-27 by skollrc