linuxboot/heads-wiki #43: Not able to sign into Slack with gmail account.

Thrilleratplay, 2020-08-27:
I tried to register to the Open Source Firmware Slack channel using a gmail account.  When trying to "Connect with Google", I am sent a verification for my 2FA but will say that my email and password are not valid.  The only a google.com domain email address can be used when creating an account.  There is a message that says

> Don’t have an @google.com email address?
Contact the workspace administrator at Open Source Firmware for an invitation.

How does one do this?

flammit, 2020-08-27:
You can get an invite at https://slack.osfw.dev/

Thrilleratplay, 2020-08-27:
That method says I already have an account.  I click the link that says "forgot password" and submit that email.

> Something went wrong and we weren’t able to send you an email. Please try again.

Thrilleratplay, 2020-08-27:
Doh! I have to click reset password on that page not continue or press enter.  I'm an idiot.

closed 2020-08-27 by Thrilleratplay

tlaurion, 2020-08-29:
Link have been added to heads-wiki https://github.com/osresearch/heads-wiki/commit/29d8940f12215f3fba1f3b5f414a406e127efd22