linuxboot/heads-wiki #59: Document kgpe-d16 usage and boards options

tlaurion, 2020-12-08:
Things are scattered around. Some drops.
- https://github.com/osresearch/heads/pull/867 has been brought back (coreboot 4.11 with TPM 1.2 support for workstation, workstation-usb and server (last one not enough tested in current state)
- https://github.com/osresearch/heads/issues/134#issuecomment-368922440 documentation should be under wiki
- https://github.com/u-root/u-bmc/issues/133 for replacement of old an unmaintained openbmc code (part of funding path)
- Is the newer flashrom still working to flash the bmc to flash it from heads?
- No TPM2 14-1 pins boards avail? (prospection https://github.com/osresearch/heads/pull/893/)

Insurgo decided to fund KGPE-D16 upstreaming efforts. [Donations welcome over open collective.](https://opencollective.com/insurgo)

Tonux599, 2020-12-09:
@tlaurion I will get started on this hopefully sometime next week. I just really have to catch up with Uni right now.

bcHelix, 2021-02-15:
Any update on this? Looking to get Heads running on my D16 when the 16M chip shows up.

Tonux599, 2021-02-15:
> Any update on this? Looking to get Heads running on my D16 when the 16M chip shows up.

@bcHelix my bad, I am still intending too I've just been quite busy. I'll try and get a PR open sometime by the end of the month. In the meantime if your struggling with anything feel free to hit me up in Slack.

tlaurion, 2021-11-02:
[16MB SPI](https://github.com/osresearch/heads/issues/719#issuecomment-956746240): [W25Q128FVSG SOP8 chip soldered on top of a SOP8 SO8 SOIC8 TO DIP8 adapter](https://github.com/flashrom/flashrom/issues/185#issuecomment-956825657)

tlaurion, 2021-11-02:
Immunefi is funding upstreaming work: https://github.com/osresearch/heads/issues/719#issuecomment-893024079