linuxboot/heads-wiki #61: document kexec-boot

jtmoree-github-com, 2021-01-10:
I am studying kexec-boot to create documentation for it with an eye on fixing my own system.  I'm reading source but still new to heads and not understanding everything.  Hoping an issue here will get some feedback to speed up my work.  Will make a pull request when I have something significant done.

I have cloned heads and heads-wiki.  Currently working on adding -h to kexec-boot with descriptions of params.  Can someone explain how -a and -r work?  Well I see how they work but don't understand what they are trying to do?  esp. '-r'

flammit, 2021-01-10:
These two options are used to add (`-a`) or remove (`-r`) arguments in the kernel command line (which usually come from parsing the grub, syslinux, BLS files).  Anything provided to the `-a` argument is appended to the command line and every argument that matches a `-r` value is removed from the command line.

The original example is in the x230 board config:

```
export CONFIG_BOOT_KERNEL_ADD="intel_iommu=on intel_iommu=igfx_off"
export CONFIG_BOOT_KERNEL_REMOVE="quiet"
```

This would be equivalent to always calling kexec-boot with these arguments and the effect is to always remove the "quiet" option from the kernel (so we can see verbose output in the target kernel), and then enabling iommu for everything except the integrated gpu.

Hope that helps!

tlaurion, 2021-01-10:
```
while getopts "b:e:r:a:o:fi" arg; do
        case $arg in
                b) bootdir="$OPTARG" ;;
                e) entry="$OPTARG" ;;
                r) cmdremove="$OPTARG" ;;
                a) cmdadd="$OPTARG" ;;
                o) override_initrd="$OPTARG" ;;
                f) dryrun="y"; printfiles="y" ;;
                i) dryrun="y"; printinitrd="y" ;;
```

Current use cases:
```
user@localhost:~/heads$ grep -Rn "kexec-boot" ./initrd/bin/
./initrd/bin/kexec-select-boot:299:		INITRD=`kexec-boot -b "$bootdir" -e "$option" -i` \
./initrd/bin/kexec-select-boot:308:		kexec-boot -b "$bootdir" -e "$option" \
./initrd/bin/kexec-select-boot:312:		kexec-boot -b "$bootdir" -e "$option" -a "$add" -r "$remove" \
./initrd/bin/oem-factory-reset:258:    ( cd /boot && /bin/kexec-boot -b "/boot" -e "$entry" -f \
./initrd/bin/kexec-save-default:125:cd $bootdir && kexec-boot -b "$bootdir" -e "$entry" -f | \
```

By looking into kexec-select-boot line 312, we see that $add and $remove are added.
In normal conditions, those add and remove parameters come from board config, where the board define what parameters need to be added or removed from the OS default for that board to function as expected.
the x230 for example defines:
```
export CONFIG_BOOT_KERNEL_ADD="intel_iommu=on intel_iommu=igfx_off"
export CONFIG_BOOT_KERNEL_REMOVE="quiet"

on kexec-select-boot line 331, we can see what is pushed as kernel parameters to have a red screen when the user selects to boot a boot selection without validating its content against detached signed digest (called force here):
```
# Allow a way for users to ignore warnings and boot into their systems
# even if hashes don't match
        if [ "$force_boot" = "y" ]; then
                scan_options
                # Remove boot splash and make background red in the event of a forced boot
                add="$add vt.default_red=0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff"
                remove="$remove splash quiet"
                user_select
        fi

So if board config specifies quiet, but we go into non-verified mode, add sets red background, while remove statement removes splash and quiet, so that we are in verbose mode when we boot our OS.

Does that answer your question? Thanks for this documenting effort @jtmoree-github-com 
```
```

tlaurion, 2021-01-10:
haha @flammit that was made at same time! Thanks for documenting :)

jtmoree-github-com, 2021-01-10:
Yes, I am incorporating these answers into the docs I am building.  I can boot a system using kexec but not when I try to use kexec-boot.  This works with either --append or --comand-line set for the root dev
`kexec -l /boot/vmlinuz --initrd=/boot/initrd.img --append='root=/dev/sdaX'`
but this does not work.  
`kexec-boot -b /boot/vmlinuz -o /boot/initrd.img -e 'root=/dev/sdaX'`
It complains about missing the intel_iommu=on so I put it in but it still does not work.
`kexec-boot -b /boot/vmlinuz -o /boot/initrd.img -e 'root=/dev/sdaX intel_iommu=on intel_iommu=igfx_off'`
Am I using kexec boot incorrectly?

flammit, 2021-01-10:
The format of the entry argument to pass to `-e` is described here in Installing-and-Configuring/install-os.md line 246.

The equivalent kexec-boot command to your kexec command would be:

`kexec-boot -b /boot -e "description|elf|kernel /vmlinuz|initrd /initrd.img|append root=/dev/sdaX"`

jtmoree-github-com, 2021-01-11:
created a pull request https://github.com/osresearch/heads/pull/960