linuxboot/heads-wiki #63: Add glossary

Thrilleratplay, 2021-01-24:
Many of the terms in Heads can quickly become confusing, example "key".

mutablescutoid, 2022-06-25:
Would an separate page in the wiki be sufficient for now? I don't have enough experience with Jekyll to implement any sort of plugin.

tlaurion, 2022-06-25:
Glossary would probably an additional section of https://osresearch.net/Keys/

For heads-wiki contribution, please read the following https://osresearch.net/Contributing-to-Heads-wiki/

mutablescutoid, 2022-06-26:
A more general glossary might benefit the newbies described in Thrilleratplay's comment in #25; I'm free to work on it if manpower is a concern

tlaurion, 2022-06-26:
@mutablescutoid sorry, I clicked your link to issue #25 and I'm not sure what glossary part you are talking about in this issue. Maybe some quote or addition to issue #25? Which is pretty long and also misleading at this point?

I would love tonknow what you are talking about without having to read the whole issue or try tongueas what you are talking about. Can you be more clear?

mutablescutoid, 2022-06-26:
I was talking specifically about https://github.com/osresearch/heads-wiki/issues/25#issuecomment-769982877