linuxboot/heads-wiki #69: dead links in wiki

cl445, 2021-03-12:
All internal links at least at the index.md are pointing to nowhere (404).

Thrilleratplay, 2021-03-12:
@cl445  Are you linking through the markdown file in this repo or navigating through https://osresearch.net?  The links within the markdown files are intended to processed through Jekyll and hosted, not browsed within the github repo.

cl445, 2021-03-15:
Yes, this what I did unintentionally. I understand that this could not work. 
The fact that confuse me is that the README.md from the heads repository is pointing to the index.md in this repo. 
Wouldn't it be a better idea to change the link to a working version? Or at least not to link to the Index.md but only to the repository?

Thrilleratplay, 2021-03-15:
@cl445  Agreed and didn't realize that link existed.  I put in a [PR to change that](https://github.com/osresearch/heads/pull/986).  Sorry for the confusion.

closed 2021-03-15 by tlaurion