linuxboot/heads-wiki #70: Document how to investigate reproducibility issues

tlaurion, 2021-05-24:
Expend from https://github.com/osresearch/heads/issues/927#issuecomment-744075679

tlaurion, 2021-05-25:
Then to compare local to CircleCI artifacts, you need to download and extract what was packaged (initrd.cpio.xz) and compare it with local build in case of mismatch of rom images signatures.

[Origin](https://github.com/osresearch/heads/issues/590#issuecomment-623143332)

We are interested here into commit https://github.com/osresearch/heads/commit/9339394b699fb1790eea3623d6ad5acf7334297a (9339394 commit id. Note: appended into rom name).
- CircleCI builds happens on https://app.circleci.com/pipelines/github/osresearch/heads
- Look for build for 9339394. Look at the jobs for your board. Click on it ([here x230-htop-maximized](https://app.circleci.com/pipelines/github/osresearch/heads/286/workflows/056d69aa-89d7-4278-80ec-b51c7738efbb/jobs/361) for sake of example).
- Click on the artifacts tab, which brings you [here](https://app.circleci.com/pipelines/github/osresearch/heads/286/workflows/056d69aa-89d7-4278-80ec-b51c7738efbb/jobs/361/artifacts)
- This is the content of `buid/x230-htop-maximized` directory. For the sake of the reproducibility issues that happens in binaries we build, it will always be encompassed into the initrd.cpio.xz file (compressed bundled tools.cpio, modules.cpio, heads.cpio). 

Now lets compare:

Remote:
```
mkdir -p ~/Downloads/new-test
wget https://361-64810881-gh.circle-artifacts.com/0/build/x230-hotp-maximized/initrd.cpio.xz -O ~/Downloads/new-test/initrd.cpio.xz
cd ~/Downloads/new-test
xz -d initrd.cpio.xz ; for i in initrd.cpio modules.cpio tools.cpio heads.cpio; do cpio -i < $i; done

```
Local:
```
mkdir -p new-test
cp initrd.cpio.xz new-test/
cd new-test
xz -d initrd.cpio.xz ; for i in initrd.cpio modules.cpio tools.cpio heads.cpio; do cpio -i < $i; done
```
Will edit with diffoscope insight.

tlaurion, 2021-05-25:
veritysetup difference between 2 debian-10 hosts with different paths of builds:

```
user@x230-test:~/heads$ diffoscope  ~/Downloads/new-test/bin/veritysetup build/x230-hotp-maximized/new-test/bin/veritysetup 
--- /home/user/Downloads/new-test/bin/veritysetup
+++ build/x230-hotp-maximized/new-test/bin/veritysetup
├── readelf --wide --sections {}
│ @@ -3,16 +3,16 @@
│  Section Headers:
│    [Nr] Name              Type            Address          Off    Size   ES Flg Lk Inf Al
│    [ 0]                   NULL            0000000000000000 000000 000000 00      0   0  0
│    [ 1] .interp           PROGBITS        0000000000400200 000200 000019 00   A  0   0  1
│    [ 2] .hash             HASH            0000000000400220 000220 000254 04   A  4   0  8
│    [ 3] .gnu.hash         GNU_HASH        0000000000400478 000478 000038 00   A  4   0  8
│    [ 4] .dynsym           DYNSYM          00000000004004b0 0004b0 000780 18   A  5   1  8
│ -  [ 5] .dynstr           STRTAB          0000000000400c30 000c30 0003f9 00   A  0   0  1
│ -  [ 6] .gnu.version      VERSYM          000000000040102a 00102a 0000a0 02   A  4   0  2
│ +  [ 5] .dynstr           STRTAB          0000000000400c30 000c30 0003fc 00   A  0   0  1
│ +  [ 6] .gnu.version      VERSYM          000000000040102c 00102c 0000a0 02   A  4   0  2
│    [ 7] .gnu.version_r    VERNEED         00000000004010d0 0010d0 000040 00   A  5   2  8
│    [ 8] .rela.dyn         RELA            0000000000401110 001110 000048 18   A  4   0  8
│    [ 9] .rela.plt         RELA            0000000000401158 001158 0006f0 18  AI  4  21  8
│    [10] .init             PROGBITS        0000000000401848 001848 00000d 00  AX  0   0  1
│    [11] .plt              PROGBITS        0000000000401860 001860 0004b0 10  AX  0   0 16
│    [12] .text             PROGBITS        0000000000401d10 001d10 002d71 00  AX  0   0 16
│    [13] .fini             PROGBITS        0000000000404a81 004a81 000008 00  AX  0   0  1
├── readelf --wide --dynamic {}
│ @@ -2,28 +2,28 @@
│  Dynamic section at offset 0x6e08 contains 26 entries:
│    Tag        Type                         Name/Value
│   0x0000000000000001 (NEEDED)             Shared library: [libcryptsetup.so.4]
│   0x0000000000000001 (NEEDED)             Shared library: [libuuid.so.1]
│   0x0000000000000001 (NEEDED)             Shared library: [libdevmapper.so.1.02]
│   0x0000000000000001 (NEEDED)             Shared library: [libpopt.so.0]
│   0x0000000000000001 (NEEDED)             Shared library: [libc.so]
│ - 0x000000000000000f (RPATH)              Library rpath: [//lib:/root/project/install/lib]
│ + 0x000000000000000f (RPATH)              Library rpath: [//lib:/home/user/heads/install/lib]
│   0x000000000000000c (INIT)               0x401848
│   0x000000000000000d (FINI)               0x404a81
│   0x0000000000000004 (HASH)               0x400220
│   0x000000006ffffef5 (GNU_HASH)           0x400478
│   0x0000000000000005 (STRTAB)             0x400c30
│   0x0000000000000006 (SYMTAB)             0x4004b0
│ - 0x000000000000000a (STRSZ)              1017 (bytes)
│ + 0x000000000000000a (STRSZ)              1020 (bytes)
│   0x000000000000000b (SYMENT)             24 (bytes)
│   0x0000000000000015 (DEBUG)              0x0
│   0x0000000000000003 (PLTGOT)             0x607000
│   0x0000000000000002 (PLTRELSZ)           1776 (bytes)
│   0x0000000000000014 (PLTREL)             RELA
│   0x0000000000000017 (JMPREL)             0x401158
│   0x0000000000000007 (RELA)               0x401110
│   0x0000000000000008 (RELASZ)             72 (bytes)
│   0x0000000000000009 (RELAENT)            24 (bytes)
│   0x000000006ffffffe (VERNEED)            0x4010d0
│   0x000000006fffffff (VERNEEDNUM)         2
│ - 0x000000006ffffff0 (VERSYM)             0x40102a
│ + 0x000000006ffffff0 (VERSYM)             0x40102c
│   0x0000000000000000 (NULL)               0x0
├── readelf --wide --version-info {}
│ @@ -1,10 +1,10 @@
│  
│  Version symbols section '.gnu.version' contains 80 entries:
│ - Addr: 000000000040102a  Offset: 0x00102a  Link: 4 (.dynsym)
│ + Addr: 000000000040102c  Offset: 0x00102c  Link: 4 (.dynsym)
│    000:   0 (*local*)       2 (CRYPTSETUP_1.0)    0 (*local*)       0 (*local*)    
│    004:   0 (*local*)       0 (*local*)       0 (*local*)       0 (*local*)    
│    008:   0 (*local*)       2 (CRYPTSETUP_1.0)    3 (LIBPOPT_0)     0 (*local*)    
│    00c:   3 (LIBPOPT_0)     0 (*local*)       0 (*local*)       0 (*local*)    
│    010:   3 (LIBPOPT_0)     3 (LIBPOPT_0)     0 (*local*)       2 (CRYPTSETUP_1.0) 
│    014:   0 (*local*)       2 (CRYPTSETUP_1.0)    2 (CRYPTSETUP_1.0)    2 (CRYPTSETUP_1.0) 
│    018:   2 (CRYPTSETUP_1.0)    0 (*local*)       2 (CRYPTSETUP_1.0)    0 (*local*)
├── readelf --wide --decompress --hex-dump=.dynstr {}
│ @@ -58,10 +58,10 @@
│    0x00400fa0 61646473 65740073 6967656d 70747973 addset.sigemptys
│    0x00400fb0 65740073 69677072 6f636d61 736b0073 et.sigprocmask.s
│    0x00400fc0 69676163 74696f6e 00737472 6572726f igaction.strerro
│    0x00400fd0 725f7200 7365746c 6f63616c 65005f5f r_r.setlocale.__
│    0x00400fe0 6c696263 5f737461 72745f6d 61696e00 libc_start_main.
│    0x00400ff0 4c494250 4f50545f 30004352 59505453 LIBPOPT_0.CRYPTS
│    0x00401000 45545550 5f312e30 002f2f6c 69623a2f ETUP_1.0.//lib:/
│ -  0x00401010 726f6f74 2f70726f 6a656374 2f696e73 root/project/ins
│ -  0x00401020 74616c6c 2f6c6962 00                tall/lib.
│ +  0x00401010 686f6d65 2f757365 722f6865 6164732f home/user/heads/
│ +  0x00401020 696e7374 616c6c2f 6c696200          install/lib.

```

tlaurion, 2021-05-25:
cryptsetup difference between 2 debian-10 hosts with different paths of builds:

user@x230-test:~/heads$ diffoscope  ~/Downloads/new-test/bin/cryptsetup build/x230-hotp-maximized/new-test/bin/cryptsetup
--- /home/user/Downloads/new-test/bin/cryptsetup
+++ build/x230-hotp-maximized/new-test/bin/cryptsetup
├── readelf --wide --sections {}
│ @@ -3,16 +3,16 @@
│  Section Headers:
│    [Nr] Name              Type            Address          Off    Size   ES Flg Lk Inf Al
│    [ 0]                   NULL            0000000000000000 000000 000000 00      0   0  0
│    [ 1] .interp           PROGBITS        0000000000400200 000200 000019 00   A  0   0  1
│    [ 2] .hash             HASH            0000000000400220 000220 000358 04   A  4   0  8
│    [ 3] .gnu.hash         GNU_HASH        0000000000400578 000578 000038 00   A  4   0  8
│    [ 4] .dynsym           DYNSYM          00000000004005b0 0005b0 000ac8 18   A  5   1  8
│ -  [ 5] .dynstr           STRTAB          0000000000401078 001078 0006e5 00   A  0   0  1
│ -  [ 6] .gnu.version      VERSYM          000000000040175e 00175e 0000e6 02   A  4   0  2
│ +  [ 5] .dynstr           STRTAB          0000000000401078 001078 0006e8 00   A  0   0  1
│ +  [ 6] .gnu.version      VERSYM          0000000000401760 001760 0000e6 02   A  4   0  2
│    [ 7] .gnu.version_r    VERNEED         0000000000401848 001848 000040 00   A  5   2  8
│    [ 8] .rela.dyn         RELA            0000000000401888 001888 000048 18   A  4   0  8
│    [ 9] .rela.plt         RELA            00000000004018d0 0018d0 000a38 18  AI  4  21  8
│    [10] .init             PROGBITS        0000000000402308 002308 00000d 00  AX  0   0  1
│    [11] .plt              PROGBITS        0000000000402320 002320 0006e0 10  AX  0   0 16
│    [12] .text             PROGBITS        0000000000402a00 002a00 005e81 00  AX  0   0 16
│    [13] .fini             PROGBITS        0000000000408881 008881 000008 00  AX  0   0  1
├── readelf --wide --dynamic {}
│ @@ -2,28 +2,28 @@
│  Dynamic section at offset 0xce08 contains 26 entries:
│    Tag        Type                         Name/Value
│   0x0000000000000001 (NEEDED)             Shared library: [libcryptsetup.so.4]
│   0x0000000000000001 (NEEDED)             Shared library: [libpopt.so.0]
│   0x0000000000000001 (NEEDED)             Shared library: [libuuid.so.1]
│   0x0000000000000001 (NEEDED)             Shared library: [libdevmapper.so.1.02]
│   0x0000000000000001 (NEEDED)             Shared library: [libc.so]
│ - 0x000000000000000f (RPATH)              Library rpath: [//lib:/root/project/install/lib]
│ + 0x000000000000000f (RPATH)              Library rpath: [//lib:/home/user/heads/install/lib]
│   0x000000000000000c (INIT)               0x402308
│   0x000000000000000d (FINI)               0x408881
│   0x0000000000000004 (HASH)               0x400220
│   0x000000006ffffef5 (GNU_HASH)           0x400578
│   0x0000000000000005 (STRTAB)             0x401078
│   0x0000000000000006 (SYMTAB)             0x4005b0
│ - 0x000000000000000a (STRSZ)              1765 (bytes)
│ + 0x000000000000000a (STRSZ)              1768 (bytes)
│   0x000000000000000b (SYMENT)             24 (bytes)
│   0x0000000000000015 (DEBUG)              0x0
│   0x0000000000000003 (PLTGOT)             0x60d000
│   0x0000000000000002 (PLTRELSZ)           2616 (bytes)
│   0x0000000000000014 (PLTREL)             RELA
│   0x0000000000000017 (JMPREL)             0x4018d0
│   0x0000000000000007 (RELA)               0x401888
│   0x0000000000000008 (RELASZ)             72 (bytes)
│   0x0000000000000009 (RELAENT)            24 (bytes)
│   0x000000006ffffffe (VERNEED)            0x401848
│   0x000000006fffffff (VERNEEDNUM)         2
│ - 0x000000006ffffff0 (VERSYM)             0x40175e
│ + 0x000000006ffffff0 (VERSYM)             0x401760
│   0x0000000000000000 (NULL)               0x0
├── readelf --wide --version-info {}
│ @@ -1,10 +1,10 @@
│  
│  Version symbols section '.gnu.version' contains 115 entries:
│ - Addr: 000000000040175e  Offset: 0x00175e  Link: 4 (.dynsym)
│ + Addr: 0000000000401760  Offset: 0x001760  Link: 4 (.dynsym)
│    000:   0 (*local*)       2 (CRYPTSETUP_1.0)    2 (CRYPTSETUP_1.0)    0 (*local*)    
│    004:   0 (*local*)       0 (*local*)       0 (*local*)       2 (CRYPTSETUP_1.0) 
│    008:   0 (*local*)       0 (*local*)       0 (*local*)       2 (CRYPTSETUP_1.0) 
│    00c:   3 (LIBPOPT_0)     0 (*local*)       3 (LIBPOPT_0)     0 (*local*)    
│    010:   0 (*local*)       0 (*local*)       3 (LIBPOPT_0)     2 (CRYPTSETUP_1.0) 
│    014:   2 (CRYPTSETUP_1.0)    3 (LIBPOPT_0)     0 (*local*)       2 (CRYPTSETUP_1.0) 
│    018:   0 (*local*)       2 (CRYPTSETUP_1.0)    2 (CRYPTSETUP_1.0)    0 (*local*)
├── readelf --wide --decompress --hex-dump=.dynstr {}
│ @@ -104,11 +104,11 @@
│    0x004016c8 736f0073 7464696e 00736967 61646473 so.stdin.sigadds
│    0x004016d8 65740073 6967656d 70747973 65740073 et.sigemptyset.s
│    0x004016e8 69677072 6f636d61 736b0073 69676163 igprocmask.sigac
│    0x004016f8 74696f6e 00737472 6572726f 725f7200 tion.strerror_r.
│    0x00401708 7365746c 6f63616c 65005f5f 6c696263 setlocale.__libc
│    0x00401718 5f737461 72745f6d 61696e00 4c494250 _start_main.LIBP
│    0x00401728 4f50545f 30004352 59505453 45545550 OPT_0.CRYPTSETUP
│ -  0x00401738 5f312e30 002f2f6c 69623a2f 726f6f74 _1.0.//lib:/root
│ -  0x00401748 2f70726f 6a656374 2f696e73 74616c6c /project/install
│ -  0x00401758 2f6c6962 00                         /lib.
│ +  0x00401738 5f312e30 002f2f6c 69623a2f 686f6d65 _1.0.//lib:/home
│ +  0x00401748 2f757365 722f6865 6164732f 696e7374 /user/heads/inst
│ +  0x00401758 616c6c2f 6c696200                   all/lib.

tlaurion, 2021-05-25:
cryptsetup-reencrypt difference between 2 debian-10 hosts with different paths of builds:

```
user@x230-test:~/heads$ diffoscope  ~/Downloads/new-test/bin/cryptsetup-reencrypt build/x230-hotp-maximized/new-test/bin/cryptsetup-reencrypt
--- /home/user/Downloads/new-test/bin/cryptsetup-reencrypt
+++ build/x230-hotp-maximized/new-test/bin/cryptsetup-reencrypt
├── readelf --wide --sections {}
│ @@ -3,20 +3,20 @@
│  Section Headers:
│    [Nr] Name              Type            Address          Off    Size   ES Flg Lk Inf Al
│    [ 0]                   NULL            0000000000000000 000000 000000 00      0   0  0
│    [ 1] .interp           PROGBITS        0000000000400200 000200 000019 00   A  0   0  1
│    [ 2] .hash             HASH            0000000000400220 000220 000294 04   A  4   0  8
│    [ 3] .gnu.hash         GNU_HASH        00000000004004b8 0004b8 000038 00   A  4   0  8
│    [ 4] .dynsym           DYNSYM          00000000004004f0 0004f0 000900 18   A  5   1  8
│ -  [ 5] .dynstr           STRTAB          0000000000400df0 000df0 0004f0 00   A  0   0  1
│ -  [ 6] .gnu.version      VERSYM          00000000004012e0 0012e0 0000c0 02   A  4   0  2
│ -  [ 7] .gnu.version_r    VERNEED         00000000004013a0 0013a0 000060 00   A  5   3  8
│ -  [ 8] .rela.dyn         RELA            0000000000401400 001400 000048 18   A  4   0  8
│ -  [ 9] .rela.plt         RELA            0000000000401448 001448 000870 18  AI  4  21  8
│ -  [10] .init             PROGBITS        0000000000401cb8 001cb8 00000d 00  AX  0   0  1
│ +  [ 5] .dynstr           STRTAB          0000000000400df0 000df0 0004f3 00   A  0   0  1
│ +  [ 6] .gnu.version      VERSYM          00000000004012e4 0012e4 0000c0 02   A  4   0  2
│ +  [ 7] .gnu.version_r    VERNEED         00000000004013a8 0013a8 000060 00   A  5   3  8
│ +  [ 8] .rela.dyn         RELA            0000000000401408 001408 000048 18   A  4   0  8
│ +  [ 9] .rela.plt         RELA            0000000000401450 001450 000870 18  AI  4  21  8
│ +  [10] .init             PROGBITS        0000000000401cc0 001cc0 00000d 00  AX  0   0  1
│    [11] .plt              PROGBITS        0000000000401cd0 001cd0 0005b0 10  AX  0   0 16
│    [12] .text             PROGBITS        0000000000402280 002280 0044f1 00  AX  0   0 16
│    [13] .fini             PROGBITS        0000000000406771 006771 000008 00  AX  0   0  1
│    [14] .rodata           PROGBITS        0000000000406780 006780 001990 00   A  0   0 32
│    [15] .eh_frame_hdr     PROGBITS        0000000000408110 008110 00013c 00   A  0   0  4
│    [16] .eh_frame         PROGBITS        0000000000408250 008250 0008cc 00   A  0   0  8
│    [17] .ctors            PROGBITS        0000000000608dd0 008dd0 000010 00  WA  0   0  8
├── readelf --wide --symbols {}
│ @@ -90,10 +90,10 @@
│      86: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND sigaction
│      87: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND memcpy
│      88: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND strlen
│      89: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND crypt_get_dir@CRYPTSETUP_1.0 (2)
│      90: 0000000000000000     0 FUNC    GLOBAL DEFAULT  UND getline
│      91: 0000000000608df8     8 OBJECT  GLOBAL DEFAULT   19 stdout
│      92: 0000000000608df0     8 OBJECT  GLOBAL DEFAULT   19 stderr
│ -    93: 0000000000401cb8     1 FUNC    GLOBAL DEFAULT   10 _init
│ +    93: 0000000000401cc0     1 FUNC    GLOBAL DEFAULT   10 _init
│      94: 0000000000608e00     8 OBJECT  GLOBAL DEFAULT   19 stdin
│      95: 0000000000406771     1 FUNC    GLOBAL DEFAULT   13 _fini
├── readelf --wide --relocs {}
│ @@ -1,15 +1,15 @@
│  
│ -Relocation section '.rela.dyn' at offset 0x1400 contains 3 entries:
│ +Relocation section '.rela.dyn' at offset 0x1408 contains 3 entries:
│      Offset             Info             Type               Symbol's Value  Symbol's Name + Addend
│  0000000000608df0  0000005c00000005 R_X86_64_COPY          0000000000608df0 stderr + 0
│  0000000000608df8  0000005b00000005 R_X86_64_COPY          0000000000608df8 stdout + 0
│  0000000000608e00  0000005e00000005 R_X86_64_COPY          0000000000608e00 stdin + 0
│  
│ -Relocation section '.rela.plt' at offset 0x1448 contains 90 entries:
│ +Relocation section '.rela.plt' at offset 0x1450 contains 90 entries:
│      Offset             Info             Type               Symbol's Value  Symbol's Name + Addend
│  0000000000609018  0000000100000007 R_X86_64_JUMP_SLOT     0000000000000000 crypt_keyslot_status@CRYPTSETUP_1.0 + 0
│  0000000000609020  0000000200000007 R_X86_64_JUMP_SLOT     0000000000000000 memcmp + 0
│  0000000000609028  0000000300000007 R_X86_64_JUMP_SLOT     0000000000000000 ioctl + 0
│  0000000000609030  0000000400000007 R_X86_64_JUMP_SLOT     0000000000000000 write + 0
│  0000000000609038  0000000500000007 R_X86_64_JUMP_SLOT     0000000000000000 tcsetattr + 0
│  0000000000609040  0000000600000007 R_X86_64_JUMP_SLOT     0000000000000000 sscanf + 0
├── readelf --wide --dynamic {}
│ @@ -2,28 +2,28 @@
│  Dynamic section at offset 0x8e08 contains 26 entries:
│    Tag        Type                         Name/Value
[... 669 lines cut ...]
│    403f25:	41 5c                	pop    %r12
│    403f27:	c6 05 f2 ac 20 00 01 	movb   $0x1,0x20acf2(%rip)        
│    403f2e:	5d                   	pop    %rbp
│    403f2f:	c3                   	retq   
│    403f30:	c3                   	retq   
│    403f31:	66 66 2e 0f 1f 84 00 	data16 nopw %cs:0x0(%rax,%rax,1)
│ @@ -1705,15 +1705,15 @@
[... cut ...]
│ -  403f58:	e8 a3 c0 bf ff       	callq  0 <_init@@Base-0x401cb8>
│ +  403f58:	e8 a3 c0 bf ff       	callq  0 <_init@@Base-0x401cc0>
│    403f5d:	5d                   	pop    %rbp
│    403f5e:	e9 0d ff ff ff       	jmpq   403e70 <getline@plt+0x1c00>
│    403f63:	0f 1f 44 00 00       	nopl   0x0(%rax,%rax,1)
│    403f68:	e9 03 ff ff ff       	jmpq   403e70 <getline@plt+0x1c00>
│    403f6d:	0f 1f 00             	nopl   (%rax)
│    403f70:	41 57                	push   %r15
│    403f72:	31 c0                	xor    %eax,%eax

```

tlaurion, 2021-05-25:
libblkid.so.1 difference between 2 debian-10 hosts with different paths of builds is too big to output here.

tlaurion, 2021-05-25:
busybox difference between 2 debian-10 hosts with different paths of builds is too big to output here.

mainly, time of compilation is injected, all address of code is shifted, including entry point.

tlaurion, 2021-05-25:
@osresearch @flammit : any troubleshooting insights would be amazing for the community. Otherwise I'll try to fix when I have time...

tlaurion, 2021-05-27:
> @tlaurion I don't know how precisely CI looks like, but you know my rambling about reproducible builds. The only feasible paths are Nix/Guix related and having reproducible build environment. This moves very slow because everyone is short on time. Anyway Dasharo has this on roadmap and have to address the issue.I gathered some thoughts to have one point to which I can reference people when reproducibility problem arise: https://docs.dasharo.com/osf-trolling-list/build_process/

@pietrushnic This seems to exactly be the problem here, with cryptsetup and busybox being reproducible in the past but having broke reproducibility recently, where I do not quite understand without further digging. What happened and why RPATHs are now introduced in the cryptsetup binaries and libs, where they weren't before :
```
│ - 0x000000000000000f (RPATH)              Library rpath: [//lib:/root/project/install/lib]
│ + 0x000000000000000f (RPATH)              Library rpath: [//lib:/home/user/heads/install/lib]
```

As for coreboot layer, those are now again reproducible (was not in the past, relaying on musl-cross-make, now relaying on coreboot buildstack per coreboot version) which is now "pinned" per version of coreboot; specified in the board configuration.

So yeah. We have all libraries and binaries outside of coreboot being built against musl-cross-make and local host toolchain. Seems like host toolchain introduced changes.... again.

CI is over [here](https://circleci.com/gh/osresearch/heads) based on current [CircleCI config](https://github.com/osresearch/heads/blob/master/.circleci/config.yml):
- Pulls [debian-10 docker](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/.circleci/config.yml#L37)
- [installs dedian-10 host tools](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/.circleci/config.yml#L40-L43): `build-essential zlib1g-dev uuid-dev libdigest-sha-perl libelf-dev bc bzip2 bison flex git gnupg iasl m4 nasm patch python python2 python3 wget gnat cpio ccache pkg-config cmake libusb-1.0-0-dev autoconf texinfo ncurses-dev doxygen graphviz udev libudev1 libudev-dev automake libtool rsync innoextract`
- Starts heads build process per board (while reusing cache if existing, out of scope here)
  - builds [make](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/Makefile#L641-L683) and [gawk](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/Makefile#L53-L102) if version discreteness, then [musl-cross-make](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/Makefile#L161-L181). All other modules depends on [CROSS_COMPILE](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/modules/linux#L46) [which definitions are under global Makefile ](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/Makefile#L166-L184)
  - Starts building [coreboot](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/Makefile#L193-L199), including its buildstack first from [config in board config](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/boards/qemu-coreboot/qemu-coreboot.config#L4-L6), where coreboot depends on [its own, version specific, buildstack](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/modules/coreboot#L59-L68)
  - Builds linux headers (headers installs) with musl-cross-make (**which configure depends on host tools**)
  - Builds board specific requirements (totp-tools, cryptsetup, gpg2, etc) from muls-cross-make and **leaked configure dependencies**
  - Wraps previous dev.cpio, tools.cpio, heads.cpio into [initrd.cpio.xz](https://github.com/osresearch/heads/blob/214703f92dff45032cb5c82ada777eb42a297f32/Makefile#L530-L611); builds coreboot rom.

tlaurion, 2021-07-08:
@pietrushnic Innteresting! https://github.com/NixOS/nixpkgs/pull/128592

tlaurion, 2021-07-21:
Notes on resolving reproducibility issues, just because some questions were asked over slack and wanted to keep track
- [rpath patches that were applied to other libs and binaries in the past](https://github.com/osresearch/heads/search?q=rpath)

pietrushnic, 2021-07-21:
Yes, I know @felixsinger works on that. Please note there will be stable toolchains of coreboot, based on recent decisions of coreboot leadership. More to that there would be stability checks also. So things going in very good direction. Please check notes:
https://docs.google.com/document/d/1NRXqXcLBp5pFkHiJbrLdv3Spqh1Hu086HYkKrgKjeDQ/edit?usp=sharing

tlaurion, 2021-07-22:
> Yes, I know @felixsinger works on that. Please note there will be stable toolchains of coreboot, based on recent decisions of coreboot leadership. More to that there would be stability checks also. [...]

For anyone else interested: see notes from 30 June 2021 forward

tlaurion, 2022-02-24:
Redid the test as part of [passing O2->Os](https://github.com/osresearch/heads/pull/1121) (in the goal of reducing modules footprint at the cost of performance)

The list of non-reproducible binaries grows.

Here is the small script I came up with, having a local board build under ~/heads while having dumped CircleCI hashes.txt for the same board under a local directory (/tmp/circleci_cryptsetup_for_all here):
`rm -f ./output_non_reproducible ./identical_hashes ./bad_hashes; grep / hashes.txt | awk -F " " '{print $1}'| while read hash; do grep -q $hash ~/heads/build/x230-hotp-maximized/hashes.txt; if [ $? == 0 ]; then echo $hash >> ./identical_hashes; else echo $hash >> ./bad_hashes;fi;done; cat ./bad_hashes| while read bad; do grep $bad hashes.txt >> ./output_non_reproducible; done;`

cat ./output_non_reproducible shows:
```
b52b5c56ddd870a88c5fd58cc78259afa85b5a9dfad3d86f5df935d1dd668c3c  /root/project/build/x230-hotp-maximized/tools.cpio
0926dde50a4ded3286f445ed515f3c9e27771022b153fe63cb8c3be597cd0391  ./lib/libcryptsetup.so.12
dcb42777d0ab5ee22be9abe9b834c20ea12cfd5ee1d493b664e7c3d1a56308d6  ./lib/libjson-c.so.5
4e27c1681daf9882c05d7a8f1c617593ee0e100cf4aad1e55f1833dd91b44545  ./lib/libgcrypt.so.20
5cee77d4ae3ee96e6a6b6856302449ffad4ff6691414946b53f71fa614e93345  ./lib/libtpm.so
d648d4b6710774bc16b8d50183614061d4bb22eace15c97b99853e4747006a13  ./lib/libuuid.so.1
f521557a2616a2fccd8738827526bdc77aade773a3b24ba627543ef0e4c8af1d  ./lib/libblkid.so.1
d634a36cc7cf24d7b201c385c1fecf0d0faa39761944c3398e671fc208bc06fd  ./bin/busybox
d9858fcbebd3ceb34473f3d05dbd4be4f745f36276516434ece24f1fccc5a387  ./bin/totp
e6d46749fe3bd9af195d614f867f4bade0d9a32505635e06cdf574a1ae635fe1  ./bin/hotp
1e285a68eb9e978bea867e87fabbc1e1288d78495cd8e0f938a16330c81c9f2e  ./bin/base32
d1204ecf1df92d7feadfc47f13be1f6c9d8a8a78dd22cda3f8668c145121bf2b  ./bin/qrenc
ca39155972ec33a70cf0cbaca56486a003a0e01be22595d1bbf9aff6aeeb22af  ./bin/tpm
1e117685faf7b1b28dc43b538d45af3f1e95fe8dbaaa41fbff23b9a84b264cbc  ./bin/cryptsetup
7f9b14e92370c3a26688e7372cdd59e8dba9c6db460f9486c45cdcd6c291c445  ./bin/cryptsetup-reencrypt
86feb33089789978b41a58007b603c7aef7309a64986cac38a84e648fdd81875  ./bin/veritysetup
efc02e76ba8fabbe47dbfd7ae86a7f4026b3f12cbdd44272fbe5241823311602  ./bin/dmsetup
d6eb80e207a1397f0d7185692942e9e3e12ddd91ae2b6f35e3fb469869ec24fa  ./bin/lvm
da4f0c6eea7861f3c5709a9eb066a34db2d2114f993f43276731601d753d2f29  ./bin/fbwhiptail
5178f615a6a5e1604061a4d7a599a1590b1ed51c151fec8dfca004293d08bb7c  build/x230-hotp-maximized/initrd.cpio.xz

```

Where the following pinpoints to the possible modules being responsible to produce such binaries:
`cat output_non_reproducible | awk -F "/" '{print $NF}'| while read bin; do grep $bin -R ~/heads/modules|awk -F ":" '{print $1}'; done|sort|uniq`

Which produces the following (not completely valid) list:
```
/home/user/heads/modules/busybox
/home/user/heads/modules/coreboot
/home/user/heads/modules/cryptsetup
/home/user/heads/modules/cryptsetup2
/home/user/heads/modules/fbwhiptail
/home/user/heads/modules/gawk
/home/user/heads/modules/hidapi
/home/user/heads/modules/hotp-verification
/home/user/heads/modules/json-c
/home/user/heads/modules/libgcrypt
/home/user/heads/modules/linux
/home/user/heads/modules/linuxboot
/home/user/heads/modules/lvm2
/home/user/heads/modules/qrencode
/home/user/heads/modules/tpmtotp
/home/user/heads/modules/util-linux
```

The curated version of output_non_reproducible file produces following modules now being non-reproducible modules:
`user@localhost:/tmp/circleci_cryptsetup_for_all$ cat output_non_reproducible | awk -F "/" '{print $NF}'| while read bin; do grep $bin -R ~/heads/modules|awk -F ":" '{print $1}'; done|sort|uniq|awk -F "/" '{print $NF}'`
- busybox
- cryptsetup
- cryptsetup2
- fbwhiptail
- hidapi
- hotp-verification
- json-c
- libgcrypt
- lvm2
- qrencode
- tpmtotp
- util-linux

tlaurion, 2022-02-25:
@jans23 @mrchromebox : do we do plan together to resolve this sharing knowledge we have in the area to bring back reproducible builds?

I may have access to grant money to fix this and would love to know who is the most knowledgeable to fix those

tlaurion, 2022-02-25:
And would also gladly and proactively document such process in the contribute section and also learn my way with such paid mentorship.

osresearch, 2022-03-30:
One technique I've found useful is to have a unique string in the build path and then grep through the various outputs for it.  This found that json-c was including some paths for the musl headers, and that in turn causes cryptsetup2 to be irreproducible.

cryptsetup1 ignores the `--disable-rpath` flag to configure, so it burns rpaths into the binaries.  cryptsetup2's patch forcibly removes that code, so it would need to be back ported.

busybox should be fixed by #74 or by adding `-DBB_EXTRA_VERSION='"none"'` to the build flags so that it doesn't use the autoconf timestamp.

haven't gotten to the others yet

osresearch, 2022-03-30:
One of my thoughts for ensuring reproducibility is to have two CI runners with different base OS images and require that they produce the same firmware images for any commits.  That will double the CPU required, although it is important to catch these early rather than after things have bit-rotted or when someone asks "hey, when did this break?".

osresearch, 2022-04-02:
A few other important things:
* Linux must have an initrd provided, or else it will generate a irreproducible one from the host system's `/dev/console`
* Linux 4.x seems to include some full paths, so it is necessary to pass `-ffile-prefix-map=....` to filter them out.
* The cross compilers also need to have their paths rewritten; this was causing issues with cairo
* zlib tries to detect cross compilers and ends up breaking some of the tools; only built the libraries to avoid it
* coreboot's `.config` file can end up with absolute paths for the bzimage and initrd, which will then mess up the hashes if `CONFIG_INCLUDE_CONFIG_FILE=y` is set since the `.config` will be included in the ROM.

osresearch, 2022-04-08:
Anything using wild cards is suspect... it looks like lvm2 might have a problem in its makefile that globs the exported symbols: https://github.com/osresearch/linux-builder/issues/6

tlaurion, 2023-04-11:
As of today, https://github.com/osresearch/heads/pull/1367 fixed json-c.

Here are the current reproducibility at the moment of merging that PR (when using fbwhiptail):
```
user@heads-tests:/tmp$ HASHES1=~/heads/build/x86/x230-hotp-maximized/hashes.txt
user@heads-tests:/tmp$ HASHES2=/tmp/hashes.txt
user@heads-tests:/tmp$ egrep '^[0-9a-f]{64}' $HASHES1 | while read line; do HASH_REF=$(echo $line|awk -F " " {'print $1'}); FILE_REF=$(echo $line|awk -F "/" {'print $NF'}); if ! grep -q "$HASH_REF" $HASHES2; then echo "$FILE_REF doesn't match";fi; done
tools.cpio doesn't match
libmbedcrypto.so.0 doesn't match
libgcrypt.so.20 doesn't match
fbwhiptail doesn't match
kexec doesn't match
busybox doesn't match
bash doesn't match
initrd.cpio.xz doesn't match
heads-x230-hotp-maximized-v0.2.0-1508-g46aa253-bottom.rom doesn't match
heads-x230-hotp-maximized-v0.2.0-1508-g46aa253.rom doesn't match
```

saper, 2023-04-14:
So, I have managed to overcome [my debian 11 issues](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=983357#205https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=983357#205) and it seems that I was able to reproduce [libgcrypt build from CI](https://app.circleci.com/pipelines/github/saper/heads/6/workflows/ee96c813-24e4-4529-ac8c-f809e489fd07/jobs/211/artifacts) on a freshly deployed Debian 11 amd64  virtual machine: 

```
$ find . -name libgcrypt* | while read X; do sha256sum "$X"; done
5cabab0f0523fae5a6e50adbf217bc3d67ade4efde0f4d6a828189d1942f19d4  ./build-unpack/lib/libgcrypt.so.20
5cabab0f0523fae5a6e50adbf217bc3d67ade4efde0f4d6a828189d1942f19d4  ./ci-unpack/lib/libgcrypt.so.20
```