linuxboot/heads-wiki #83: T430 w530 flashing notes

tlaurion, 2021-12-07:
@eganonoa wrote : "Note: it can be tricky to properly read the 8MB chip on the T530 and W530 (see #996). This appears to be because the two chips share a power source on these boards (see generally). This is true for both the Rasbperry Pi and the CH431a. There appears to be two workarounds: (1) the Wake on Lan method (which has the one downside of only working once, because it is not supported by coreboot and is, therefore, not available after flashing); or (2) keeping the CMOS battery plugged in, which works also post-flashing. Both methods provide a little extra power to the board and appear to then be sufficient to get enough power to the 8MB chip to get a proper read of it. The Skulls team generally suggest using an external power source with the CH341a, and it's possible that this might help, but I have not tested."

From https://github.com/osresearch/heads/pull/1072#issue-1071482504

This needs to be documented properly

eganonoa, 2021-12-11:
> @eganonoa wrote : "Note: it can be tricky to properly read the 8MB chip on the T530 and W530 (see #996). This appears to be because the two chips share a power source on these boards (see generally). This is true for both the Rasbperry Pi and the CH431a. There appears to be two workarounds: (1) the Wake on Lan method (which has the one downside of only working once, because it is not supported by coreboot and is, therefore, not available after flashing); or (2) keeping the CMOS battery plugged in, which works also post-flashing. Both methods provide a little extra power to the board and appear to then be sufficient to get enough power to the 8MB chip to get a proper read of it. The Skulls team generally suggest using an external power source with the CH341a, and it's possible that this might help, but I have not tested."
> 
> From [osresearch/heads#1072 (comment)](https://github.com/osresearch/heads/pull/1072#issue-1071482504)
> 
> This needs to be documented properly

@tlaurion Will work on it. Should be able to make progress on various wiki-related things over the holidays.

I'm still learning the idiosyncrasies of these 15" boards. My latest finding is that the WAL or CMOS battery workarounds are not necessary if you use the wonderful blue LC Technology CH341A programmer (e.g. like this one [here](https://fluxworkshop.com/products/beaa100020-ch341a-usb-to-uart-i2c-spi-ttl-isp-epp-mem-lc-blue)).  That gets a clean read on the 8MB chip T530 and W530 boards without either workarounds and unlike the black CH341A, which appears ubiquitous these days) does not send 5v through the data channels. So, my latest assumption is that this the safest and most efficient way to work on the t530 and w530 boards. Nonetheless, I'll document all three options when I work on this.

tlaurion, 2022-07-03:
https://github.com/osresearch/heads/issues/996#issuecomment-1172985391

For another way of flashing without disassembling the whole laptop?

3hhh, 2022-07-03:
From my experience with coreboot and two T530:

I physically cut out the part of the board case that was blocking access to the ROM chips. This is a bit archaic, but works if one's not clumsy. I used a relatively strong side cutter. Pomona worked fine afterwards.

For flashing with a RPi I followed these instructions:
        1. remove the AC adapter & all batteries
                Lenovo BIOS: remove the CMOS battery
                bricked BIOS or Coreboot: leave the CMOS battery attached
        2. attach Pomona clip incl 5 wires for CS -> CE0, SO -> MISO, SI -> MOSI, SCLK -> SCLK, GND -> GND (do NOT connect Vcc, HOLD and WP seem optional as the motherboard appears to drive them to 3V3)
        3. attach the AC adapter (not battery!), press Power button & wait for the laptop to boot
                if it doesn't (CPU fan doesn't start):
                        a make sure the CMOS battery is attached (the following won't work otherwise)
                        b unplug the AC adapter
                        c press the power button and hold it multiple times in a row (empty capacitors: should perform a hard reset)
                        d attach the AC adapter
                        e it should power now...
                        f Note: external PSU was tested and doesn't work
        4. keep the BIOS ROM powered:
                Lenovo BIOS: laptop states that there's a md5sum error with the CMOS (the CMOS battery is removed after all)
                Coreboot/SeaBios: wait for Seabios to pop up, press Esc and never select a boot device
        5. measure CS on the chip - if it's constant 3V3 apparently no one else is driving the chip (might make more sense to do that in a first run without wires attached)
        6. power up the Pi and read multiple times, compare the images afterwards, optionally write
        7. power down the pi, then the laptop (for writing: do not reboot the laptop, but do the first chip, power down the pi, re-attach the Pamona, power up the pi and continue with the other chip)

I.e. usually I had to internally power the flash chips.

A comment from another user that I recall:
"There are no pull-up resistors on CS#. Connecting CS# of the opposite chip to the 3.3V from the RaspberryPi helped in my case."

Internal flashing is really useful for that one later on as physical can be a bit messy (see above). So I hope that heads doesn't disable the internal flashing option.

tlaurion, 2022-07-03:
@3hhh seems like 3.3v is just not enough to reliably have a reading of the 8mb spi. Tests as in OP here suggests to have CMOS battery connected to have reliable reading.

Another path for flashing without disassembling is superficially highlighted here for w530 https://github.com/hamishcoleman/thinkpad-ec/issues/70#issuecomment-374564806

Someone has more info on that soldering pad? Pictures?

tlaurion, 2022-07-03:
Some pictures
https://www.coreboot.org/images/thumb/5/5d/T530_SPI_chips.jpg/300px-T530_SPI_chips.jpg

Cn100 pad:
https://duckduckgo.com/?q=w530+CN100&iar=images&iax=images&ia=images&iai=https%3A%2F%2Fwww.elvikom.pl%2Fpictures%2F00047237.jpg

Schematics:
https://duckduckgo.com/?q=w530+CN100&iar=images&iax=images&ia=images&iai=https%3A%2F%2Fi.imgur.com%2FSH87v8V.png

Someone played with that?

nomenkai, 2023-05-31:
I am confused. The title mentions T430 but I find no mention of it within this thread. Am I missing something here?

tlaurion, 2023-06-02:
@nomenkai I'm not sure if t430 is part of this documentation issue and I think it was in issue title per error.

T530 and W530 are from same series and shared same issues with underpowered/overpowered programmer.

I am removing t430 of title.

tlaurion, 2023-06-02:
It is to note that ch341a 1.6+ versions have a switch for voltage which mitigate the issue just like the referred LC technology version.

As opposed to that specific version, ch431a 1.6 programmers have a switch.

I would recommend buying from 3mdeb shop for a complete kit with wires preassembled.

fhvyhjriur, 2024-05-13:
Ebay and other sources now have a 'ch341a v1.7' that have a switch. They are widely available. Here a search-link: https://www.ebay.com/sch/i.html?_nkw=ch341a+v1.7