linuxboot/heads-wiki #102: Provisioning a new Yubikey to be used with Heads

githubuseravailable, 2022-09-11:
hi @tlaurion 

i see the Heads wiki has been improved a lot than before, and it is great. 

do you mind to add steps about how to provision a new Yubikey and export its public key ? 
or maybe there is any good references about that ? 

do we have to choose between using the original private key in the Yubikey, 
or generate key, then store it inside Yubikey ?

tlaurion, 2022-09-11:
Is there any reason why https://osresearch.net/Configuring-Keys/#no-public-key-found-in-rom is not answering your question?

Which Yubikey please ? ( please modify issue title so that others find this issue).

Know that you might face https://github.com/osresearch/heads/issues/1076

Let us know

tlaurion, 2022-09-11:
I suggested a workaround there, but the user who opened issue didn't follow through, and I do not own a yubikey myself.

https://github.com/osresearch/heads/issues/1076#issuecomment-1146270706

githubuseravailable, 2022-09-11:
okay, i have changed the issue title, is it good now, or maybe modify again ?

i also don't really understand that part https://osresearch.net/Configuring-Keys/#no-public-key-found-in-rom

You can either: 
- add a backuped gpg public key matching an already provisioned USB Security dongle (see below on adding public key) or 
- generate the keys, alongside as setting all security components in one go with the OEM Factory Reset/Re-Ownership option.

does it mean that, if we choose OEM factory reset/re-ownersip, 
then we don't need to provision the Yubikey ? 

or OEM factory reset/re-ownership, is the next step after add gpg key to the running bios ?

tlaurion, 2022-09-11:
Basically, I would love to know if forsedsig is on :

gpg --card-status 

Then we could go for an image to test and fix issue.

Since Heads bonds more nicely to nitrokey pro and librem keys, no other bug report having happened and the discussion having stopped over there left the issue unfixed.

Please try oem-factory reset/Re-ownwrship wizard and report back in the same issue/PR to advance the discussion. Tag me there as well and I'll try to fix code for you to test image.

Which board?

Upload

tlaurion, 2022-09-11:
> You can either:
>
> add a backuped gpg public key matching an already provisioned USB Security dongle (see below on adding public key) or
> generate the keys, alongside as setting all security components [...]

The OEM factory reset/re-ownership provisions the USB security dongle if it is in factory state, or Re-owns states, including keypair with your personal information.

What would need to be clearer? Without a keypair (private key in smartcard, public key exported) with a Realname, email address and optinal comment, and passphrases, the dongle cannot really be used.

githubuseravailable, 2022-09-11:
> gpg --card-status

Signature PIN: not forced
Signature, encryption, authentication, general key: none

> Which board?

x230

> Please try oem-factory reset/Re-ownwrship wizard and report back in the same issue/PR to advance the discussion. Tag me there as well and I'll try to fix code for you to test image.

okay

> How would you put that in its simplest form so you would have understood that from quoted text?

okay, i understand better now, thanks

i thought that the Yubikey will have contained original private key without provisioning before. 
maybe if users don't understand GPG and Yubikey, it will be hard to understand this part. 

maybe if we modify the wiki, it will be better, such as: 

note that the factory state USB security dongle contain no key inside,
it need to be provisioned to contain key, 

if your USB security dongle is in factory state, it means it has not been provisioned, 
then choose OEM factory reset, to provision it. 

but if you already have the provisioned USB security dongle, and the backuped gpg public key, 
then can choose add gpg key to the running bios

githubuseravailable, 2022-09-14:
@tlaurion i have followed the solution in [oem factory reset fail](https://github.com/osresearch/heads/issues/1076#issuecomment-1146270706)
and Heads configuration has completed now, thanks

but there are other issues, so i will open ticket

tlaurion, 2022-09-14:
@githubuseravailable 

> @tlaurion i have followed the solution in https://github.com/osresearch/heads/issues/1076#issuecomment-1146270706
> and Heads configuration has completed now, thanks

So basically, you confirm that the code is not dealing properly with Yubikey 5 Nano, and that https://github.com/osresearch/heads/issues/1076#issuecomment-1146270706 is stil lvalid and that a code fix is needed?

githubuseravailable, 2022-09-17:
> So basically, you confirm that the code is not dealing properly with Yubikey 5 Nano, and that [osresearch/heads#1076 (comment)](https://github.com/osresearch/heads/issues/1076#issuecomment-1146270706) is stil lvalid and that a code fix is needed?

@tlaurion  i didn't try oem factory reset without forcesig, 
i assume directly that it will fail without forcesig, 
so then gpg card edit, admin, forcesig directly, 
do you want me to re-try oem factory reset without forcesig ? 
i can re-try if you think we need the experiment

githubuseravailable, 2022-10-01:
ok @tlaurion i will try without forcesig to ensure, then apply the additional script above, 
i will update again, but not at anytime soon.

tlaurion, 2022-10-02:
That should be implemented first and tested. 

Reopening issue

closed 2023-02-21 by tlaurion