linuxboot/heads-wiki #103: RTC clock drifting problems - manual/automatic correction guides

tlaurion, 2022-09-14:
Not sure how that happeneds and from where, but I have found no reference under osresearch.net on ntp/skew/time/date to automatically set/manually correct time after first flash.

There is also no actual reference to network-recovery-init to sync time against router through NTP protocol, nor manual instructions on fixing the time in case of https://github.com/osresearch/heads/issues/1021 (which is most probably a bad RTC clock battery, needing to be changed, but how users can know if not documented).

Basically, an entree should be aded under heads-wiki so that users can easily find what is documented under https://github.com/osresearch/heads/issues/1021#issuecomment-1158982540

githubuseravailable, 2022-09-15:
@tlaurion 

maybe it's part of configuration, since most of 2nd hand device, will have bad RTC battery, 
but at the same time, it's also an issue. 

so, maybe it's good idea to document it under heads-wiki, 
for heads issue already documented in [1021](https://github.com/osresearch/heads/issues/1021)
with link to this ticket, under heads-wiki

tlaurion, 2022-09-15:
https://github.com/osresearch/heads-wiki/pull/78 covers that, needs review and adaptation.

Fix fourth issue reported https://github.com/osresearch/heads/issues/1213#issuecomment-1248412613

githubuseravailable, 2022-09-18:
@tlaurion 

i have tried this 1st solution, in [TOTP Mismatch](https://github.com/osresearch/heads-wiki/pull/78)

date -s hh:mm:ss && hwclock -w

but it will set the HW clock only, without setting the HW date,
therefore the HW date still incorrect,
maybe that's the reason why the TOTP still mismatch,
even after setting the HW clock correctly

githubuseravailable, 2022-09-19:
@tlaurion   i tried 

date -s 'yyyy-mm-dd hh:mm:ss' && hwclock -w

and now the date time has been set correctly, 
and TOTP finally match, 
maybe can add on into the wiki

tlaurion, 2023-01-13:
Should have not been closed. PR should be modified accordingly to code changes upstream.

tlaurion, 2023-01-13:
Sorry for the noise. PR that needs to be modified is here: https://github.com/osresearch/heads-wiki/pull/78

closed 2023-01-13 by tlaurion