linuxboot/heads-wiki #110: Reflashing Heads - Internal vs External

newbieAtGithub, 2022-12-05:
hi @tlaurion 

which one provide better integrity, internal or external Heads re-flashing ?

thanks & regards,

tlaurion, 2022-12-06:
> which one provide better integrity, internal or external Heads re-flashing ?

External is better, but unneeded in most cases?

Can you detail your concern? I think I answered this partly under references given under other #111 #112 #113 #114 issues you opened targeting different angles of the same question, which all goes to: if you build yourself, and that you can verify that got checkout is clean, that ROM is not "dirty" and that you flash and reflash the same firmware externally and internally with same TOTP/HOTP valid attestation, and TPM disk encryption key passphrase releases the encryption key and boots the final OS correctly, then the firmware being reflashed keeping those valid should reassure you.

newbieAtGithub, 2022-12-09:
hi @tlaurion 

> Can you detail your concern?

my concern is, if we are unsure, about the integrity of the flashed image, 
let's say the flashed image has been compromised, 
is it still okay, to update Heads, by re-flashing update internally ? 

> External is better, but unneeded in most cases

in what case we need external re-flash ?

tlaurion, 2022-12-09:
https://osresearch.net/Updating#reflashing-the-same-firmware

Once builds will be reproducible again, any ROM build locally or on CircleCI will have the same final hash, and hashes.txt will be the same for a got commit.

Until then, reflashing thr same firmware internally should be enough to reassure most, where backup of same ROM and unpacking it to verify should also be enough.

One could verify the hash for flashrom, trusting internally that the checksum is reported correctly, meaning that busybox is not compromised. One should backup the ROM and inspect it externally against hashes.txt to make sure everything is good.

When internally flashing, flashrom is used. When checking for digest integrity, sha256sum -c is a subprogram of busybox.

Once again, the only real good option to backup and verify an untrusted ROM, if radical, would be to take a backup externally, unpack the ROM and verify its hashes to be as expected.

Reflashing externally will invalidate measurements, since the ROM won't contain the same gpg keyring. This is why reflashing internally should be the first step, after having took a backup of the ROM to be inspected.

The risks flashrom being compromised are again really low while possible. 

I would advise into asking specific questions in the issue I created as raw notes for integrity validation, which hopefully will result into an additional wiki page to answer all those questions.

newbieAtGithub, 2022-12-09:
hi @tlaurion 

okay, in short, you suggested that external re-flash is better, but not necessary, because: 
- internal flashing requires internal flashrom only
- the risk of internal flashrom being compromised is really low, although possible
- we can use busybox's sha256sum -c to verify internal flashrom
- also can internal backup the flashed rom, unpack & inspect it externally against hashes.txt
- Reflashing externally will invalidate measurements, since the ROM will contain different gpg keyring

> I would advise into asking specific questions in the issue I created as raw notes for integrity validation, which hopefully will result into an additional wiki page to answer all those questions.

okay

closed 2022-12-09 by newbieAtGithub