linuxboot/heads-wiki #111: Verifying Heads Before Compiling

newbieAtGithub, 2022-12-05:
hi @tlaurion, 

Below are some steps to verify Heads before compiling: 
1. git clone Heads from Github
2. download Heads' artifact, from latest commit & build in Circle CI
3. verify all files in cloned Heads with hashes.txt from step 2

Please kindly add on or modify the steps above, for a better verification. 

Thanks and Regards,

newbieAtGithub, 2022-12-09:
hi @tlaurion 

i refer to [hashes](https://output.circle-artifacts.com/output/job/cae4a10f-c7bc-43c3-a788-ee1c9bca7cb0/artifacts/0/build/x86/x230-maximized/hashes.txt)
i found some files, actually have same name, with files in [Heads repo](https://github.com/osresearch/heads)

for example, 
./bin/kexec are the same as /heads/modules/kexec
./bin/cryptsetup are the same as /heads/modules/cryptsetup

so, maybe we can use these checksum in hashes.txt from Circle CI, 
to verify some files, cloned from Heads' repo, before local compiling, 
although the built ROM will have different checksum, due to reproducibility issue, 

is it correct ? or maybe wrong ?

thanks and regards,

tlaurion, 2022-12-09:
modules/cryptsetup is the build recipe to build cryptsetup, same for all other modules. Those modules include expected hashes of tarballs, verified after download and prior of extraction.

Everything built leaves at trace under hashes.txt

So verifying before compiling here relies on git. git will tell you if any of the files maintained in the repository are as expected.

`git status` under heads directory will tell you what is different from the upstream repository, and will create a commit-dirty version if compiled. Dirty here would mean the ROM didn't come from a clean got repository, the ROM has been built from a repository containing changes.

newbieAtGithub, 2022-12-09:
hi @tlaurion 

i see, so it is not necessary, to do manual verification, before local compiling, 
because Heads can rely on git status, to do it automatically for us, 
& it will create commit-dirty version of ROM, in case the ROM is not clean / containing change. 

okay, understand now, 

thanks and regards,

closed 2022-12-09 by newbieAtGithub