linuxboot/heads-wiki #112: Verifying Heads after Compiling

newbieAtGithub, 2022-12-05:
hi @tlaurion,

Below are some steps to verify Heads after compiling: 
1. download Heads' artifact, from latest commit & build in Circle CI
2. build / compile Heads following the steps in https://osresearch.net/Building/
3. extract content from built ROM & Coreboot payload, 
4. verify all extracted files from step 3 with hashes.txt from downloaded Heads' artifact

For step 3 & 4 can refer to [audit flashed image](https://github.com/osresearch/heads-wiki/issues/107)

Please kindly add on or modify the steps above, for a better verification.

Thanks and Regards,

tlaurion, 2022-12-06:
Unfortunately the same answer to your other tickets. Once ROMs will be reproducible (meaning building locally and from CircleCI will produce the same exact ROM) what you are seeking to accomplish will be possible.

You imply as of today that ROM from CircleCi and locally built will he the same.

They won't since some components are not bit by bit reproducible as of now.
If a binary is built differently, one single bit difference will change multiple parts of the ROM (a tool built is in tools.cpio and then under initrs.cpio.xz and then in final ROM).  

So if one component is not reproducible, the ROM is not reproducible. There are already guides here in issues to investigate this, aimed at developers to help with the current buildsystem.

Otherwise you will have to be patient so that Heads changes its build system to rely on NixOS and produces ROMs from a reproducible docker image.

Otherwise things leak in produced ROMs, both from CircleCI (which builds under /root/project which path bleeds into busybox and other problems) that are not currently resolvable without changing the way we previously expected build tools to not impact the final ROM.

It proved to be a wrong approach. make changes, gawk changed, everything changes. Unless we pin things to create a reproducible docker image, this project will continuously struggle with this kin of issue.

I'm closing letting you search for reproducible issues.

Please reopen if you think this was closed by error and tag me.

closed 2022-12-06 by tlaurion

newbieAtGithub, 2022-12-09:
hi @tlaurion 

okay, understand now, 
Circle CI built ROM & Local built ROM will have different checksum, 

currently, maybe for temporary, 
the only way to verify local built is from its generated ROM name only, 
-dirty means some files are changed, 
-commit Id means it is clean

thanks & regards,