linuxboot/heads-wiki #113: Verifying downloaded Circle CI artifacts

newbieAtGithub, 2022-12-05:
hi @tlaurion , 

Below are some steps to verify downloaded ROM from Circle CI: 
1. download artifact from the last commit from Circle CI
2. extract content from built ROM & Coreboot payload,
3. verify all extracted files from step 2 with hashes.txt from downloaded Heads' artifact

For step 2 & 3 can refer to https://github.com/osresearch/heads-wiki/issues/107

Please kindly add on or modify the steps above, for a better verification.

Thanks and Regards,

tlaurion, 2022-12-06:
#107 is notes made to eventually create an additional entry in the wiki, rendered on https://osresearch.net 

I understand from the above that your goal would be to distrust CircleCI?

#107 is to make sure that what was flashed corresponds still to what was downloaded.

Until reproducibility issues are resolved, CircleCI hashes.txt could be used to verify Heads internally. Or as detailed there, to take a backup and extract to verify against CircleCI/built ROM and its generated hashes.txt at build time.

If you want to clarify things under #107, please quote parts there to I can modify directly. Those are notes and will not be found easily by anybody.

> download artifact from the last commit from Circle CI
> extract content from built ROM & Coreboot payload,

What you build and what you download from CircleCI won't have the same final hashes for ROMs as of today. This is documented in reproducibility issues over Heads (not heads-wiki).

Hope this is clearer.

To verify CircleCI downloaded artifacts, one can simply verify hashes of the ROMs downloaded against the ones under hashes.text

If the goal is to distrust CircleCi, as of today the only alternative is to build yourself, reproducing what CircleCI does and is explained under Building instructions, until reproducibility issues are resolved. Most of the compiled binaries match across local builds and CircleCI, but busybox if my memory is good. Since busybox is packed under tools.cpio and tools.cpio is packed under initrd.cpio.xz: busyboz, tools.cpio, initrd.cpio.xz and ROM images will have different hashes.

Please tag me if closing this issue seems an error to you.

From the title of this issue, the documentation already explains how to do this through Downloading section of the wiki

closed 2022-12-06 by tlaurion

newbieAtGithub, 2022-12-09:
hi @tlaurion 

thanks for explanation, 
okay, this ticket is closed. 

thanks and regards,