linuxboot/heads-wiki #114: Verifying Blobs (me, gbe, ifd)

newbieAtGithub, 2022-12-05:
hi @tlaurion , 

i used Flashrom to dump blobs from X230, 
then using ifdtool & me cleaner, to generate & clean the blobs (me, gbe, ifd), 
also tried the script [extract.sh](https://github.com/osresearch/heads/blob/master/blobs/xx30/extract.sh), 

but the blobs generated, have different checksum, with what's written in [x230 blobs](https://github.com/osresearch/heads/blob/master/blobs/xx30), 
also different checksum, with hashes.txt in Circle CI,
is it expected ? 

if not mistaken, each blob has unique MAC address, 
so maybe, it is expected if the checksum are different, 
but then, if they are different, then how to verify its integrity ? 

Thanks and Regards,

tlaurion, 2022-12-05:
I will close this issue considering https://github.com/osresearch/heads/blob/master/blobs/xx30/README

Should answer all your questions.

Please tag me if you think its an error

closed 2022-12-05 by tlaurion

tlaurion, 2022-12-05:
Reading your issue and understanding completely different things... You will need to clarify

> used Flashrom to dump blobs from X230,
> then using ifdtool & me cleaner, to generate & clean the blobs (me, gbe, ifd),
> also tried the script extract.sh,

I understand here that you used xx30 extract script, from a flashrom backup taken externally. This will extract ifd me and GBE from your backup and won't match hashes.txt found under xx30 blobs directory. Those are hashes of downloaded and extracted ROMs.

> but the blobs generated, have different checksum, with what's written in x230 blobs,
> also different checksum, with hashes.txt in Circle CI,
> is it expected ?

Here I am confused. Hashes from CircleCI has nothing to do directly with hashes from xx30 blobs. CircleCI uses the download and extract script, and uses those blobs to stitch the final coreboot ROM image, which is then splitted in bottom and top ROM images. Consequently I do not understand the question here.

> if not mistaken, each blob has unique MAC address,
> so maybe, it is expected if the checksum are different,
> but then, if they are different, then how to verify its integrity ?

GBE blob will contain a unique MAC if extracted. Otherwise the one in tree is generated as explained in README with a static Mac address as documented. Instructions are given to reproduce the result.

newbieAtGithub, 2022-12-09:
hi @tlaurion 

maybe the issue title is not specific enough, 
maybe a better title is, 
"verifying blobs (me, gbe, ifd), extracted from dumped / backup-ed ROM". 

but if the blobs are unique for each motherboard, 
then maybe it is pointless to verify it, 
besides, definitely Lenovo doesn't provide checksum, for each motherboard. 

> Hashes from CircleCI has nothing to do directly with hashes from xx30 blobs.

yes, after rechecking it, i realized that it doesn't contain checksum for blobs. 
before i thought, hashes.txt will contain blobs' checksum, & the blobs will be the same for all, 
which then wrong assumption.

then i think we can close the ticket. 

thanks and regards,

tlaurion, 2022-12-09:
This is output of download and extract script https://app.circleci.com/pipelines/github/osresearch/heads/473/workflows/da4c9e91-0281-47cb-a84d-067f33ec976a/jobs/5566/parallel-runs/0/steps/0-110

Ifd is coming from a sacrificed motherboard in tree, with regions modified to fit downloaded and extracted ME from latest firmware update from lenovo. GBE is a configuration blob, which was generated as explained in the README.

newbieAtGithub, 2022-12-10:
hi @tlaurion 

okay, thanks a lot for info

regards,