linuxboot/heads-wiki #115: How to boot from Heads into boot-able boot-and-nuke software ?

newbieAtGithub, 2022-12-26:
hi @tlaurion 

i created boot-able thumb drive, 
containing boot-able boot-and-nuke software, [Aban](https://aban.derobert.net/),
which is similar to [Dban](https://dban.org/). 

i tried the thumb drive, in another Laptop, which using Intel bios. 
and it can boot successfully into the software. 

Heads' boot options, can detect & display the software, 
but fail to boot into the Aban software. 
Heads can boot into any Linux OS ISO. 

please inform me, do you know how to boot into the software ?

newbieAtGithub, 2023-01-15:
hi @tlaurion ,

i have re-posted the same issue above, 
in [Heads Repo](https://github.com/osresearch/heads/issues/1279)

thanks and regards,

tlaurion, 2023-01-16:
This boots correctly without issues.
Tested on top of x230 and qemu-coreboot-fbwhiptail-tpm1 board.

Is based on Debian.

Most probable issue is a bad ISO download or a bad USB thumb drive at this point, unless ABAN is not doing the right thing in case of I/O errors for SATA drive connected.

This is tested thoroughly
- Qemu: https://github.com/osresearch/heads/issues/1279#issuecomment-1384540702
- USB thumb drive creation + test under x230-maximized : https://github.com/osresearch/heads/issues/1279#issuecomment-1384599878

Closing as irrelevant for documentation issue (works out of the box as per upstream instructions).

closed 2023-01-16 by tlaurion