linuxboot/heads-wiki #120: Add flashing "best practices"

tlaurion, 2023-01-30:
There is a lot of information out there on ch341a programmer giving too much voltage (5v vs 3.3v). Users reports of having bricked their laptops (SPI chip to motherboards resistors).

One of those example is a recent exchange on slackématrix accessible at:
https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$kcFoywTcHgKLIbRP3iJ1bgJ8vRUGZsZcCGvvJsCszgw?via=matrix.org&via=nitro.chat&via=talk.puri.sm

xx20/xx30 chips and motherboard resistors are really resilient and cheap ch341a providing 5v never caused any issue, but this may vary for other SPI chips/motherboard when doing in place programming (with a PAMONA clip connected to the chip on the motherboard.)

General guidelines, depending on motherboard is:
1. Make sure that CMOS, AC Adapter and battery is deconnected.
2. Make sure to connect the clip on CHIP prior of powering up the programmer (I never followed that even with black programmer and bricked 2 laptops to date because I didn't respect point 1 above.
3. Other guides suggest to modify ch341 programmer to make sure voltage provided is 3.3v (https://www.chucknemeth.com/laptop/lenovo-x230/flash-lenovo-x230-coreboot being one of them)
4. Other guides suggest unsoldering/resoldering the SPI for extra caution (never did that)
5. More expensive programmer (LC Technologies ch341a programmers) were suggested (good general advice)

I would invite people into discussing facts here to be used to create additional flashing best practice page to the wiki.

tlaurion, 2023-01-30:
The general guideline is to make sure that the wiring is conform to chip SPI datasheet (this is a standard generally used by a chip family. Here for xx20/xx30 and SOIC8 chips), making sure that orientation of the chip is good is mandatory. On SPI chips, there is a "dot" which inform the eye that that pin is pin 1:

1- CS / CS#
2- SO1-SIO1 / MISO
3- WP# / N/C
4- GND
5- SI-SIO0 / MOSI
6- SCLK / CLK
7- HOLD# / N/C
8- VCC

For example, those two guides inversed the picture with similar pin names:
- Datasheet : http://j5d2v7d7.stackpathcdn.com/wp-content/uploads/circuits/Macronix/25L3206E-DI1.gif
- Direct link to chucknemeth's [picture](https://www.chucknemeth.com/_media/laptop/lenovo-x230/coreboot/x230-bios-chips.jpg?w=600&tok=041472)

tlaurion, 2023-09-29:
All of this should just replace ch341a by ch341a rev 1.6+ based boards.

Example of readily available kits including proper pictures for visual identification of 1.6 board containing a voltage selector:

https://shop.3mdeb.com/shop/modules/ch341a-flash-bios-usb-programmer-kit-soic8-sop8/

tlaurion, 2023-12-18:
> 1- CS / CS#
> 2- SO1-SIO1 / MISO
> 3- WP# / N/C
> 4- GND
> 5- SI-SIO0 / MOSI
> 6- SCLK / CLK
> 7- HOLD# / N/C
> 8- VCC
> 

Attacking the w530 today. Clip that came with kit doesn't fit constrained place, needed to rewire my pamona clip with wires that came with my raspberries across the years. They are color 10 color coded, that will help me and probably others.

| SPI Chip | Pomona Clip | Adapter Board | CH341A Programmer |
| :------: | :---------: | :-----------: | :---------------: |
| Pin 1 (dot) | Pin 1 (red dot) | VCC (red) | 25XX (red stripe) |
| Pin 2 | Pin 2 | GND (black) | GND (black) |
| Pin 3 | Pin 3 | CS (yellow) | CS (yellow) |
| Pin 4 | Pin 4 | CLK (orange) | CLK (orange) |
| Pin 5 | Pin 5 | DO (blue) | DO (blue) |
| Pin 6 | Pin 6 | DI (purple) | DI (purple) |
| Pin 7 | Pin 7 (brown) | VCC or GND (brown) | NC |
| Pin 8 | Pin 8 (white) | VCC or GND (white) | NC |

Then.
1- Connect sopi6/8-dip8 daughter board on ziff connector, making sure the adapter is connected into the closest holes to you when looking at the adapt from the usb connector
[image]
[image]
2- Then do the wiring pattern above. Looks like that when done:
[image]
[image]

Will definitely glue all those together with Epoxy they slide way too easily, but that works!
Got my 8mb backup from w530. Now going to make internal flashing image that can be flashed only on that exposed 8mb chip.

tlaurion, 2023-12-30:
> W530 board owners tagged under [linuxboot/heads#692](https://github.com/linuxboot/heads/issues/692)
> 
> [w530](https://doc.coreboot.org/mainboard/lenovo/w530.html) (xx30): @eganonoa @zifxify @weyounsix (dGPU: w530-k2000m) @jnscmns (dGPU K1000M) @computer-user123 (w530 / & w530 k2000 : prefers iGPU) :
> 
> What have you done on stock bios for TPM to function correctly to setup TPM DUK? Can't seem to figure it out [n4ru/1vyrain#37 (comment)](https://github.com/n4ru/1vyrain/issues/37#issuecomment-1866575025)

Nevermind! That was https://github.com/linuxboot/heads/issues/1565

tlaurion, 2024-06-13:
That question popped again under Matrix channel https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$Y5a44cDMjH-CO96sfpn7ZCHeHkBDrsBjPEatp3mbcLE?via=matrix.org&via=nitro.chat&via=envs.net

tlaurion, 2024-06-25:
Concerns raised again at https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$n56mqz2A0ZCWP1HrqVBkRL1IGKOfqJmsGqTvI8kNP-g?via=matrix.org&via=nitro.chat&via=envs.net

tlaurion, 2024-06-27:
To be honest, I think this thread and concerns from end users are mixing two different use cases:

# Unbricking / initial flashing/non-development related dev cycles / initial external flashing was made from well testers and platform I well tested

**There I actually no need here to anything better/faster /safer then a ch341a even at 5v for most older platforms (thinkpads) if all external power sources are disconnected. **

# dev cycles / debugging /porting of new platforms / multiple flashing a day is needed / newcomers to coreboot and external flashing / insecure end users

**There are way better options then ch341a**, different prices ranges, different features across reprogrammers, including some development helper kits (Tigard best of line here) that also provide logic analyser), permits to turn programmer into a coreboot usb debugging helper (through serial over usb) which is possible through beaglebone black and Tigard. So the reason this is not yet in docs is because, depending of goals and needs, there are multiple tools out there.

perplexity.ai discussion (made public) : https://www.perplexity.ai/search/Of-all-in-nfyU4c8tQp2.2GONIONnyA

Extract of markdown table:

| Programmer          | Cost         | Theoretical Speed (16MB) | Voltage Selection | Flashrom Support | Additional Use for Debugging |
|---------------------|--------------|--------------------------|-------------------|------------------|------------------------------|
| **DediProg SF600Plus-G2** | $500-$600    | 16 seconds               | Multiple          | Not mentioned    | No                           |
| **DediProg SF100**      | $200-$300    | 22 seconds               | Multiple          | Not mentioned    | No                           |
| **CH341A**              | $5-$15       | 480 seconds (8 minutes)  | 3.3V (mod needed) | Yes              | No                           |
| **Raspberry Pi**        | $35-$80      | 16 seconds               | 3.3V (additional hardware needed for other voltages) | Yes              | No                           |
| **BeagleBone Black**    | $60-$90      | 16 seconds               | 3.3V, some pins 1.8V | Yes              | Yes                          |
| **REVELPROG-IS**        | $150-$200    | 16 seconds               | Not specified     | Not mentioned    | No                           |
| **Tigard**              | $69-$89      | 16 seconds               | 1.8V, 3.3V, 5.0V, external target voltage | Yes              | Yes                          |

tlaurion, 2024-07-02:
Will buy a Tigard and test across platforms I own. For some reason,s people don't want to comment on github and stay stubborn having sporadic and non permanent/searchable notes given on Heads matrix channel. Point against Tigard at https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$WDHLHatL2aVLFCxTK3EX9PUAo8t18yx-eufPSCHdQPk?via=matrix.org&via=nitro.chat&via=envs.net

tlaurion, 2024-07-06:
Ideally, we would be able to propose a flrefrsshed version for the beaglebone black screwdriver so that bricking could give input from testers through EHCI debugging logs, based and improved upon 
https://www.coreboot.org/BeagleBone_Black_-_screwdriver

tlaurion, 2024-07-19:
Just experienced a n41 bad rom flash from  https://github.com/linuxboot/heads/pull/1723/commits/b05fbf6d9f1b88e01ba8fe46dcaa8036c16db8f0

Points:

- Tigard : 0
- ch341a : 1

Tigard, while faster with default settings, was constantly failing on flashrom verify step, and didn't seem to even check if what was attempted to be written was actually written while verify failed at different addresses. 

Flashing with ch341a, while slower, successfully unbricked my nv41 with master rom. 

Some pictures.... 
[image]
[image]

tlaurion, 2024-08-01:
Undusted my beaglebone black, but discovered screwdriver is an unmaintained project :( 

https://www.coreboot.org/BeagleBone_Black_-_screwdriver

tlaurion, 2024-08-16:
Tigard: 1
ch341a: 1

Externally flashed x230t to test https://github.com/linuxboot/heads/pull/1541 and https://github.com/linuxboot/heads/issues/1744 in one rock two stones test. Tigard is way faster then ch341a:

```
user@localhost:~/heads$ time sudo flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 -w /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2283-g7147a60
heads-x230-maximized-v0.2.0-2283-g7147a60-bottom.rom  heads-x230-maximized-v0.2.0-2283-g7147a60-top.rom     
heads-x230-maximized-v0.2.0-2283-g7147a60.rom         heads-x230-maximized-v0.2.0-2283-g7147a60.zip         
user@localhost:~/heads$ time sudo flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 -w /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2283-g7147a60-top.rom 
flashrom unknown on Linux 6.6.42-1.qubes.fc37.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Macronix flash chip "MX25L3205(A)" (4096 kB, SPI) on ft2232_spi.
Found Macronix flash chip "MX25L3205D/MX25L3208D" (4096 kB, SPI) on ft2232_spi.
Found Macronix flash chip "MX25L3206E/MX25L3208E" (4096 kB, SPI) on ft2232_spi.
Found Macronix flash chip "MX25L3233F/MX25L3273E" (4096 kB, SPI) on ft2232_spi.
Multiple flash chip definitions match the detected chip(s): "MX25L3205(A)", "MX25L3205D/MX25L3208D", "MX25L3206E/MX25L3208E", "MX25L3233F/MX25L3273E"
Please specify which chip definition to use with the -c <chipname> option.

real	0m0.156s
user	0m0.005s
sys	0m0.009s
user@localhost:~/heads$ time sudo flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 -w /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2283-g7147a60-top.rom -c "MX25L3205D/MX25L3208D"
flashrom unknown on Linux 6.6.42-1.qubes.fc37.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Macronix flash chip "MX25L3205D/MX25L3208D" (4096 kB, SPI) on ft2232_spi.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
Thanks for your help!
Reading old flash chip contents... done.
Erasing and writing flash chip... Erase/write done.
Verifying flash... VERIFIED.

real	0m6.858s
user	0m0.003s
sys	0m0.006s

```

tlaurion, 2024-08-23:
- Tigard : 2
- Ch341a: 1

```
time sudo flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 -w  heads-nitropad-nv41-v0.2.0-2285-geb88b18.rom

flashrom unknown on Linux 6.6.21-1.qubes.fc37.x86_64 (x86_64)

flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).

Found Macronix flash chip "MX25L25635F/MX25L25645G" (32768 kB, SPI) on ft2232_spi.

===

This flash part has status UNTESTED for operations: WP

The test status of this chip may have been updated in the latest development

version of flashrom. If you are running the latest development version,

please email a report to flashrom@flashrom.org if any of the above operations

work correctly for you with this flash chip. Please include the flashrom log

file for all operations you tested (see the man page for details), and mention

which mainboard or programmer you tested in the subject line.

Thanks for your help!

Reading old flash chip contents... done.

Erasing and writing flash chip... Erase/write done.

Verifying flash... VERIFIED.

real    2m37.587s

user    0m0.007s

sys    0m0.006s
```

tlaurion, 2025-01-09:
Tigard:4
ch341:1

The previous guidelines applied makes the tigard a big winner over ch341a 1.6+ for its speed
```
false; while [ $? != 0 ]; do time sudo flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 --progress -w /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2537-gaf59704-bottom.rom -c MX25L6405D; done

flashrom v1.5.0-devel (git:v1.4.0-76-gcbdb8534) on Linux 6.6.65-1.qubes.fc37.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Found Macronix flash chip "MX25L6405D" (8192 kB, SPI) on ft2232_spi.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
You can also try to follow the instructions here:
https://www.flashrom.org/contrib_howtos/how_to_mark_chip_tested.html
Thanks for your help!
Reading old flash chip contents... done.
Updating flash chip contents... Erase/write done from 0 to 7fffff
Verifying flash... VERIFIED.

real	0m21.325s
user	0m0.000s
sys	0m0.009s
user@localhost:~/heads$ false; while [ $? != 0 ]; do time sudo flashrom -p ft2232_spi:type=2232H,port=B,divisor=4 --progress -w /home/user/heads/build/x86/x230-maximized/heads-x230-maximized-v0.2.0-2537-gaf59704-bottom.rom -c MX25L6405D; done
flashrom v1.5.0-devel (git:v1.4.0-76-gcbdb8534) on Linux 6.6.65-1.qubes.fc37.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Found Macronix flash chip "MX25L6405D" (8192 kB, SPI) on ft2232_spi.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
You can also try to follow the instructions here:
https://www.flashrom.org/contrib_howtos/how_to_mark_chip_tested.html
Thanks for your help!
Reading old flash chip contents... 
[READ: 100%]...done.
Updating flash chip contents... 
Erase/write done from 0 to 7fffff

real	0m5.326s
user	0m0.006s
sys	0m0.011s

```

21 seconds to flash 8mb chip of x230 with tigard externally.

mblanqui, 2025-02-04:
An alternative option would be to recommending a Raspberry Pi Pico over the CH431a. They're relatively inexpensive, easy to use, and far faster than the CH431a at reading and writing roms. Based on past experience, the Pico can read, write and verify a 16mb image in under 30 seconds. It requires a [serprog uf2](https://codeberg.org/Riku_V/pico-serprog) file to be copied to the Pico prior to use as a programmer, but Heads could offer a pre-compiled file (as Libreboot do) to make things easier for the end user.

[image]

tlaurion, 2025-02-04:
Interesting outcome of unverified outcome discussion with perplexity.ai :

Here is a detailed comparison of the **CH341A**, **Raspberry Pi Pico**, and **Tigard** as SPI programmers, focusing on their usability, speed, and features for flashing a 16MB SPI chip.

### Overview
Each programmer has unique strengths and weaknesses, making them suitable for different use cases. Below is a thorough breakdown:

### Comparison Table

| **Feature**                | **CH341A**                           | **Raspberry Pi Pico**               | **Tigard**                             |
|----------------------------|---------------------------------------|--------------------------------------|----------------------------------------|
| **Cost**                   | ~$2-10                               | ~$4-10                              | ~$50-70                                |
| **Speed (16MB Flash)**     | ~120 seconds                         | ~30 seconds                         | ~42 seconds [4][6]                     |
| **Voltage Levels**         | Fixed 3.3V (some models output 5V, risky) [8] | Configurable via firmware           | Configurable (1.8V, 3.3V, 5V) [2][6]  |
| **Protocols Supported**    | SPI, I2C                             | SPI (via serprog firmware)          | SPI, I2C, JTAG, SWD, UART [2][6]       |
| **Ease of Use**            | Moderate (requires drivers/software) [5] | Easy (drag-and-drop UF2 firmware)   | High (plug-and-play with FlashROM/OpenOCD) [6] |
| **Reliability**            | Moderate (prone to voltage issues) [8] | High                                | High                                   |
| **Versatility**            | Low                                  | Moderate                            | Very High                              |
| **Open Source Support**    | Limited                              | High                                | High                                   |

---

### Detailed Analysis

#### **CH341A**
- **Advantages**: Extremely cheap and widely available. Supports basic SPI flashing tasks with tools like ASProgrammer or FlashROM.
- **Disadvantages**: Slow (~120 seconds for a 16MB chip), quality varies between models (e.g., black CH341A boards often output dangerous 5V instead of 3.3V) [8]. It’s less reliable for in-place programming due to lack of voltage configuration and poor build quality [1][7].

#### **Raspberry Pi Pico**
- **Advantages**: Fast (~30 seconds for a 16MB chip), affordable, and easy to set up with serprog firmware. It’s highly reliable and provides proper voltage levels by default.
- **Disadvantages**: Limited to SPI programming unless additional firmware is developed. Slightly less versatile than Tigard.

#### **Tigard**
- **Advantages**: Robust and versatile with support for multiple protocols (SPI, I2C, JTAG, etc.) and configurable voltage levels. It’s user-friendly with native support for tools like FlashROM and OpenOCD. Reliable for both in-place programming and standalone tasks.
- **Disadvantages**: More expensive (~$50-70) than the CH341A or Pico. Slightly slower than the Pico (~42 seconds for a 16MB chip) due to protocol overhead [4][6].

---

### Recommendations
1. **For Budget Users**: The CH341A is suitable if cost is the primary concern and occasional flashing tasks are sufficient.
2. **For Speed and Simplicity**: The Raspberry Pi Pico is ideal for fast and reliable SPI programming at an affordable price.
3. **For Advanced Users**: The Tigard is the best choice for versatility and multi-protocol support despite its higher cost.

Each tool serves different needs—select based on your priorities like speed, reliability, or versatility.

Citations :
[1] Better SPI Programmer Than CH341A - Recommendations Please https://winraid.level1techs.com/t/better-spi-programmer-than-ch341a-recommendations-please/33785
[2] Building a Hardware Hacking Arsenal: The Right Bits for Every Byte https://www.guidepointsecurity.com/blog/building-a-hardware-hacking-arsenal-the-right-bits-for-every-byte/
[3] [Discussion] Tigard as a BIOS dumper/programmer - Win-Raid Forum https://winraid.level1techs.com/t/discussion-tigard-as-a-bios-dumper-programmer/102452
[4] Add in place programmer's flashing "best practices" and refer to ... https://github.com/linuxboot/heads-wiki/issues/120
[5] BIOS flash programmers. Which one is right for you? - YouTube https://www.youtube.com/watch?v=wTt4wq2Y-zs
[6] Tigard | Crowd Supply https://www.crowdsupply.com/securinghw/tigard
[7] Choosing a CH341A programmer : r/coreboot - Reddit https://www.reddit.com/r/coreboot/comments/azabhl/choosing_a_ch341a_programmer/
[8] Lenovo ThinkPad T430 / Nitrokey NitroPad T430 - Qubes OS Forum https://forum.qubes-os.org/t/lenovo-thinkpad-t430-nitrokey-nitropad-t430/5847?page=3

mblanqui, 2025-02-04:
Based on the above, the Pico sounds like a good middle ground, combining the low cost of the CH341A with the speed of the Tigard. The Pico isn't as versatile a tool as either the CH341A or the Tigard, but unlike them, it has the potential via scripting to be even more versatile than the Tigard is for a fifth of the cost. I'd imagine that the recently released Pico v2 is even faster at reading chips, but I don't have one to test with rn.

The only real downside I can see is that you have to source the clip and wires yourself separate from the Pico, but most authorised resellers of the Raspberry Pi sell the constituent parts needed to make a functional programmer.

mblanqui, 2025-02-04:
Good little comparison chart between the Pico and the Pico 2.

[image]

mblanqui, 2025-03-02:
Just learnt via the ThinkPad community Discord that the CH341a has superceded by CH347, which not only has the correct voltage, but is far faster at reading & writing chips, works with flashrom, and is just as inexpensive as its predecessor. You can get it via AliExpress here: https://www.aliexpress.com/item/1005005615745631.html

tlaurion, 2025-03-02:
> Just learnt via the ThinkPad community Discord that the CH341a has superceded by CH347, which not only has the correct voltage, but is far faster at reading & writing chips, works with flashrom, and [...]

No voltage selector as ch341a? https://github.com/linuxboot/heads-wiki/issues/120#issuecomment-1741039038

mblanqui, 2025-03-02:
> No voltage selector as ch341a 1.6+? [#120 (comment)](https://github.com/linuxboot/heads-wiki/issues/120#issuecomment-1741039038)

No, but according to [this review ](https://www.onetransistor.eu/2024/09/ch347t-programmer-schematic.html)and the associated datasheet, "it is a 3.3V only device with 5V tolerant inputs", making it safe to use.

tlaurion, 2025-03-03:
> No, but according to [this review ](https://www.onetransistor.eu/2024/09/ch347t-programmer-schematic.html)and the associated datasheet, "it is a 3.3V only device with 5V tolerant inputs", making it [...]

@mblanqui that is fine for spi chips needing 3.3v. I'm questioning that is enough for all spi chips out there and a good recommendation for users who might encounter spi chips different or not documented. A nice improvement over <1.6 ch341a, not sure it wins over spi flasher that has better speed and voltage selection for non-technical users needing to unbrick/internal initial flashing across probable multiple platforms.

Afterall, this is why this issue is still not translated into heads-wiki offial docs and still under experimentation.

As of now, I would still recommend slower ch341a 1.6+ or Tigard per all this issue content.