linuxboot/heads-wiki #147: Strange FAQ answer: Why use ancient Thinkpads instead of modern Macbooks

jonathancross, 2024-02-27:
https://github.com/linuxboot/heads-wiki/blob/master/About/FAQ.md#why-use-ancient-thinkpads-instead-of-modern-macbooks

This FAQ answer does not address the question:

> The x230 Thinkpad has coreboot support, TPM, nice keyboards and are very cheap to experiment on. Newer [Thinkpads contain Bootguard](https://mjg59.dreamwidth.org/33981.html), a closed source security function implemented by Intel to prevent unsigned custom firmware, such as coreboot and heads, from being installed.

Specifically: it doesn't mention Macbooks, but rather talks about the x230 Thinkpad.

daringer, 2024-02-28:
For me this reads like "macbook" is meant to be figuratively - representing any/most modern laptops, thus I don't think it should be explicitly mentioned...

tlaurion, 2024-03-24:
https://osresearch.net/FAQ/#why-use-ancient-thinkpads-instead-of-modern-macbooks

So we want to replace MacBook by "newer platforms" and keep example of newer thinkpads being bootguard enabled and coreboot unsupported hardware as counter examples?

@daringer : thoughts?

daringer, 2024-04-12:
I would stick with my previous comment, essentially a macbook is used figuratively here and I think it is ok like that

closed 2024-04-16 by tlaurion