linuxboot/heads-wiki #178: Create code of conduct page

tlaurion, 2025-03-06:
The past two days in matrix channel is now justifying this so that moderators have the tools to refer users to, and ban users if the code of conduct is not respected.

It seems that some users are changing alias and personas, and are now trolling the channel. 

Heads would not be the first to enforce a code of conduct, which comes with responsibilities and requirements to keep things in order. 

An attempt was done in the past with push backs. Everyone has the right to free speech, but ignoring those users in a room used for collaboration is counter productive. 

It's a fist time to me needing that tool since 2018. People come and go, but now comes the time to have the tools to keep things clear. If everyone is annoyed, without power trip involved, and user cannot self moderate itself, banning justifications are needed now. 

A good example is https://www.qubes-os.org/code-of-conduct/

tlaurion, 2025-03-06:
I was also against it, preferring free speech and auto moderation. That lead to CONTRIBUTION.md that is now under tree, merged at https://github.com/linuxboot/heads/pull/1719

ArrayBolt3, 2025-03-06:
tlaurion asked me to advise here since I mentioned I'm currently helping moderate Ubuntu's presence on Matrix.

I looked a bit at the earlier pushback mentioned to get some context. I didn't read the whole thing since my time is a bit short, but I see the primary concern was that codes of conduct lead to censorship, and censorship leads to project harm. I agree *in general* with both halves of this sentiment (yes, codes of conduct can lead to censorship, and yes, censorship can lead to project harm), but I don't think the sentiment as a whole is accurate, unless you do a bad job with your code of conduct and the enforcement thereof (and yes, projects have definitely done bad jobs at this in the past).

I'm currently helping out as a member of the Ubuntu Community Council, and help with enforcing Ubuntu's code of conduct. Ubuntu is a *massive* project at this point, so we put quite a bit of thought into how to keep the entire operation from spinning out of control. I'll try to share some of that here to see if we can come up with something that makes everyone happy.

---

First, a long bit of preamble. I'm not going to tackle the question of censorship at all here, I'm mainly just wanting to lay some groundwork and see if a CoC is even needed at this point.

First off, there are at least three different kinds of CoC-like rulesets that can be useful for a project - a proper Code of Conduct, a technical policy, and rules for communication channels. *None of these should overlap, they all have very different uses.* Generally, a CoC is used to specify social rules - how should members of the project treat each other and others when working with or representing the project? Technical policies are used for specifying technical rules - what coding style should be used, do I sign Git commits or not, how are the Git repos organized and how are the branches in them organized, etc. Communication channel rules are mostly just an etiquette guide - do we use Matrix or IRC for chat, should I wrap my emails at 72/75/78/80/whatever characters, what topic does each room have or do we even care, etc. Of these, the CoC is the trickiest to get right because it attempts to control how people behave *in general*.

The incidents that inspired this issue to be filed by tlaurion was specifically a rather intense case of what appeared to be trolling in the Matrix room. This is a combination of social and communication channel issues. It was a type of general behavior that was undesirable and damaging to the project (the social aspect), and it made the Matrix room difficult to use for others (the communication channel aspect). Right now there aren't strict rules for the Matrix room, and Heads has no code of conduct, so while it was pretty obvious to some that what was happening was bad, there wasn't any objective way to say "what's happening is bad because it broke this particular rule".

So that's the background for what happened, and the tools that can be used for keeping this kind of thing from happening so easily in the future. Now for actually talking about how to implement something that will make things work.

The first question that comes to my mind is "this was just one trolling event, do we really need to make an entire code of conduct just because one troll came in and messed things up once?" This is a really good question, because if a project makes a big move in response to an insignificant event, it can be more damaging to the project than just dismissing the insignificant event. IMO, the answer here hinges on four things:

* How bad is the event that triggered us to think about this,
* How often is it happening and do we foresee it happening more in the future, and
* Who's going to do the hard work of preventing it from happening again?
* Are people going to care when bans start happening?

If an event isn't that bad, or if it's a fluke, it may be worth just ignoring. If there's just one person doing all the moderating, a CoC also might not be needed since whatever the one moderator decides should happen is what does happen. Where a CoC comes in really handy is when you have *multiple people in charge of moderation* - if not everyone agrees on what should and shouldn't be allowed, there is going to be trouble when someone inevitably stops a behavior someone else would have allowed. Another situation they've very useful for is when the community is prone to question one's decision to ban or otherwise take disciplinary action against problematic users - it gives something objective to point to so users don't get too upset when actions have to be taken, and also allows users to give concrete feedback if someone misuses power.

Currently the answers to the above four questions are:

* As far as trolling-related issues go, the events that happened on Matrix were about as bad as they can get. It was two back-to-back instances of what I like to call "stealth trolling" - two different people came into the room with what sounded like a legitimate problem, and exhibited a pattern of behavior that was extraordinarily difficult to work around or avoid getting upset when witnessing, but was just reasonable enough to make one doubt if they should really be stopped from engaging in the room. Quoting myself from near the end of the exchange (link for those who are interested: https://matrix.to/#/!pAlHOfxQNPXOgFGTmo:matrix.org/$_NbFVD-tchVWVblmRkH8-N0wIYPNlZaqD354g6FFfOw?via=matrix.org&via=nitro.chat&via=tchncs.de):

[... cut ...]
I would also add the examples of unacceptable behavior and the scope section from the [Contributor Covenant Code of Conduct](https://www.contributor-covenant.org/version/2/1/code_of_conduct/), those are very useful. Finally, I would add some stuff about how the CoC is intended to be enforced at the end (I don't really like the Contributor Covenant's enforcement guidelines, in particular the "Warning" section could cause more trouble than it's worth with restricting people's ability to communicate).

There's probably a bit more fleshing out that could be done, but those are my ideas so far. And, wow, this comment got long.

mblanqui, 2025-03-06:
At it's core, Heads is a community, a collective endeavor, and a highly democratic one at that; but communities without rules, rights and responsibilities don't last very long. There has to be a framework for engagement, otherwise lawlessness prevails.

As a relatively new member of the community, I support the introduction of a code of conduct/rights & responsibilities; not because I support the suppression of free speech, far from it, but because I think without one freedom of speech and thought within the project will inevitably die. People will become bitter by the constant bad faith engagement by others and either implement harsh measures in an undemocratic fashion or give up on the cause entirely. Slowly over time the project's contributor list will grow smaller and smaller, til finally, it either doesn't exist at all, or is a one person dictatorship - benevolent or otherwise.

It was really hard to watch what went on in the Matrix channel yesterday, I felt particularly bad for @tlaurion who went out of his way to help the troll even when it was clear there was no helping them. If it was frustrating to watch, it must have been even more frustrating to engage with. People deserve respect, but once it's clear they don't or are unwilling to reciprocate that respect, then a line has to be drawn in the sand.

tlaurion, 2025-05-26:
Not sure if this should belong in heads-wiki (to be referred to https://osresearch.net rendered static pages) or under heads main repo as CODE_OF_CONDUCT.md under #1977

hanetzer, 2025-06-28:
https://github.com/LadybirdBrowser/ladybird/blob/master/CODE_OF_CONDUCT.md
here's a COC that's worth a damn.